thliebig/openEMS-Project #3: Missing dependencies?

jeffsf, 2016-03-22:
In trying to install using `brew install --HEAD https://raw.github.com/thliebig/openEMS-Project/master/brew/openEMS.rb` I am finding missing dependencies (at the `vtk` dependency point), including:
- sip
- pyqt

This may be a problem with the vtk brew recipe.

thliebig, 2016-03-22:
Maybe brew updated vtk to version 7. Until yesterday that was not taken care of in openEMS. I just now updated this repo too to reflect this. Maybe you just try again?

jeffsf, 2016-03-22:
Yes, brew is now using v7
```
==> Installing openEMS dependency: vtk
==> Downloading http://www.vtk.org/files/release/7.0/VTK-7.0.0.tar.gz
```

The build succeeds now, thank you! 
(I did configure vtk previously with --with-qt --with-qt5)

```
$ brew install --HEAD https://raw.github.com/thliebig/openEMS-Project/master/brew/openEMS.rb
######################################################################## 100.0%
==> Cloning https://github.com/thliebig/openEMS-Project.git
Updating /Library/Caches/Homebrew/openEMS--git
==> Checking out branch master
==> cmake . -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/openEMS/HEAD -DCMAKE_BUILD_TYPE=Release -DCMAKE_F
==> make
🍺  /usr/local/Cellar/openEMS/HEAD: 246 files, 4.3M, built in 2 minutes 11 seconds

```

thliebig, 2016-03-22:
Well thats sounds good, closing the issue then. 
About the _--with-qt5_, is it necessary to put it in the brew file?

closed 2016-03-22 by thliebig

jeffsf, 2016-03-22:
OpenEMS does build without adding --with-qt --with-qt5
I'm puzzled as to why it re-installed vtk from source, but at least OpenEMS built! 
I now have to learn how to use it...

Thanks again!

```
$ brew uninstall openEMS
Uninstalling /usr/local/Cellar/openEMS/HEAD... (246 files, 4.3M)

$ brew install vtk
==> Installing vtk from homebrew/science
==> Downloading https://homebrew.bintray.com/bottles-science/vtk-7.0.0.el_capitan.bottle.1.tar.gz
Already downloaded: /Library/Caches/Homebrew/vtk-7.0.0.el_capitan.bottle.1.tar.gz
==> Pouring vtk-7.0.0.el_capitan.bottle.1.tar.gz
==> Caveats
Even without the --with-qt option, you can display native VTK render windows
from python. Alternatively, you can integrate the RenderWindowInteractor
in PyQt, PySide, Tk or Wx at runtime. Read more:
    import vtk.qt4; help(vtk.qt4) or import vtk.wx; help(vtk.wx)

==> Summary
🍺  /usr/local/Cellar/vtk/7.0.0: 3,015 files, 107.6M

$ brew install --HEAD https://raw.github.com/thliebig/openEMS-Project/master/brew/openEMS.rb
######################################################################## 100.0%
==> Installing dependencies for openEMS: vtk
==> Installing openEMS dependency: vtk
==> Downloading http://www.vtk.org/files/release/7.0/VTK-7.0.0.tar.gz
Already downloaded: /Library/Caches/Homebrew/vtk-7.0.0.tar.gz
==> cmake -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/vtk/7.0.0 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_F
==> make
==> make install
==> Caveats
Even without the --with-qt option, you can display native VTK render windows
from python. Alternatively, you can integrate the RenderWindowInteractor
in PyQt, PySide, Tk or Wx at runtime. Read more:
    import vtk.qt4; help(vtk.qt4) or import vtk.wx; help(vtk.wx)

==> Summary
🍺  /usr/local/Cellar/vtk/7.0.0: 3,188 files, 107.8M, built in 22 minutes 48 seconds
==> Installing openEMS
==> Cloning https://github.com/thliebig/openEMS-Project.git
Updating /Library/Caches/Homebrew/openEMS--git
==> Checking out branch master
==> cmake . -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/openEMS/HEAD -DCMAKE_BUILD_TYPE=Release -DCMAKE_F
==> make
🍺  /usr/local/Cellar/openEMS/HEAD: 246 files, 4.2M, built in 1 minute 12 seconds

```