thliebig/openEMS-Project #6: stl import problem

michal777, 2019-01-14:
Hi
I tried to import stl 3D file to openems and got an error
[bug_stl.txt](https://github.com/thliebig/openEMS-Project/files/2756786/bug_stl.txt)

The openEMS complains that it can't find my stl file but QCSXCAD displayed it (so it's there).

thliebig, 2019-01-14:
You should post such issues on the openEMS forum please. Especially since that is most likely a user error. My hint, please use a full path to the stl file instead of just the name itself. openEMS and QCSXCAD otherwise search the file at different current working directories and thus do or do not find it.

michal777, 2019-01-15:
With full path it works :) Thanks for the answer and sorry for posting in wrong place.

closed 2019-01-15 by michal777