thliebig/openEMS-Project #10: Missing dependencies

horror-vacui, 2020-01-08:
Some packages seems to be outdated in the installation requirement on Xubuntu 18.04.3 LTS. See code below:
```
~ $ sudo apt-get install build-essential cmake git libhdf5-dev libvtk5-dev libboost-all-dev libcgal-dev libtinyxml-dev libqt4-dev libvtk5-qt4-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libvtk5-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libvtk7-dev:i386 libvtk6-dev:i386 libvtk7-dev libvtk6-dev

Package libvtk5-qt4-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libvtk5-dev' has no installation candidate
E: Package 'libvtk5-qt4-dev' has no installation candidate
```

Also there is a problem with tinyxml dependency. OpenEMS is not bulit. The log reveals the error message:
```
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find TinyXML (missing: TinyXML_LIBRARY TinyXML_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindTinyXML.cmake:65 (find_package_handle_standard_args)
  CMakeLists.txt:90 (find_package)
```
The issue was not solved by manually installing - multiple versions of - tinyxml:
```
 /tmp/openEMS $ dpkg -l | grep tinyxml
ii  libtinyxml-dev:amd64                                             2.6.2-4                                             amd64        TinyXml library - header and static library
ii  libtinyxml2-6:amd64                                              6.0.0+dfsg-1                                        amd64        C++ XML parsing library
ii  libtinyxml2-dev:amd64                                            6.0.0+dfsg-1                                        amd64        TinyXML2 library - header and static library
ii  libtinyxml2.6.2v5:amd64                                          2.6.2-4                                             amd64        C++ XML parsing library

```

xcoder123, 2021-03-04:
I'm having the same issue. Were libtinyxml is not being found. On ubuntu 20.04 LTS

thliebig, 2022-12-31:
I think I recently successfully tried to install in a Ubuntu 22.04 VM. Is this still an issue? Installing "libtinyxml-dev" should be sufficent!

0xCoto, 2022-12-31:
Shouldn't be an issue - been working fine on Ubuntu 20.04 and 22.04 (clean installations) with no issues whatsoever for me.

closed 2022-12-31 by thliebig