thliebig/openEMS-Project #11: Installtion on Ubuntu 20 (Focal Fossa)

p-j-r, 2020-06-14:
Several of the dependencies are not available on the latest Ubuntu. Please suggest the replacements for the repositories:
`sudo apt-get install build-essential cmake git libhdf5-dev libvtk5-dev libboost-all-dev libcgal-dev libtinyxml-dev libqt4-dev libvtk5-qt4-dev`
_libvtk5-dev_ has been replaced by libvtk7-dev; Libraries **libqt4-dev** and **libvtk5-qt4-dev** are not available.

jsr38, 2020-06-25:
try `qt5base-dev` and `libvtk7-qt-dev` in place of `libqt4-dev` and `libvtk5-qt4-dev` respectively.  Note that I can't vouch for a fulling working build as a result but it'll get you on the way.

thliebig, 2020-06-26:
I don't have or use Ubuntu these days. Thus I cannot check, but it sounds about right. openEMS should be used with Qt5 and vtk 7 or 8 (9 should work too)

closed 2020-06-27 by p-j-r