thliebig/openEMS-Project #12: Mistype?

yrabbit, 2020-06-25:
```
% fgrep -Ri Primtive
openEMS/python/openEMS/ports.py:    :param priority: int -- priority of all contained primtives
CSXCAD/python/CSXCAD/CSPrimitives.pyx:        Create a primtive specified by the `prim_type`
CSXCAD/src/CSPrimitives.h:      bool GetPrimitiveUsed() {return m_Primtive_Used;}
CSXCAD/src/CSPrimitives.h:      void SetPrimitiveUsed(bool val) {m_Primtive_Used=val;}
CSXCAD/src/CSPrimitives.h:      bool m_Primtive_Used;
CSXCAD/src/CSPrimitives.cpp:    m_Primtive_Used = false;
```

thliebig, 2020-06-26:
Looks like it. Not really an issue as it is either internal or just a comment?

yrabbit, 2020-06-26:
Well, yes, but it’s not that I ran a spell check :) I figured out unused primitives.

closed 2020-06-26 by yrabbit