thliebig/openEMS-Project #13: CMakeLists.txt missing CTB and hyp2mat

nvonwolf, 2021-02-01:
Would it be possible to include CTB/hyp2mat in the CMakeLists.txt like you have done for AppCSXCAD/QCSXCAD?

I am trying to build the project using the [Spack](https://spack.io/) packaging system (common in High Performance Computing environments).  Packaging using cmake is relatively straight forward but relying on a bash script instead highly complicates the install.

thliebig, 2021-02-01:
I think the CTB is included in openEMS?
hyp2mat is pretty much a stand-alone tool that does not use cmake (afaik) and you would need to ask the other of the tool to change that.

nvonwolf, 2021-02-01:
Cool I can package hyp2mat separately so that's no big deal,  I just wasn't sure how tied into this project it was.  On another note  are you no longer doing release versions?  I see the last one was in 2017.

thliebig, 2021-02-01:
Well the current (master) version seems to be quite stable and could be a release candidate, I just don't find the time and motivation to do release testing. And I somehow always wanted to improve the documentation before I do that..

nvonwolf, 2021-02-01:
Thanks for the info.  I'll go ahead and close this ticket.

closed 2021-02-01 by nvonwolf