thliebig/openEMS-Project #15: Undefined reference to 'GDALDriver::CreateCopy...'

obstinacyvalue, 2022-01-21:
I am trying to build openEMS on an Arch Linux variant Garuda and I keep getting this error when trying to compile from scratch:

/usr/bin/ld: /usr/lib/libvtkGeovisGDAL.so.1: undefined reference to `GDALDriver::CreateCopy(char const*, GDALDataset*, int, char**, int (*)(double, char const*, void*), void*)'
/usr/bin/ld: /usr/lib/libvtkGeovisGDAL.so.1: undefined reference to `GDALDriver::Create(char const*, int, int, int, GDALDataType, char**)'
collect2: error: ld returned 1 exit status
make[5]: *** [CMakeFiles/AppCSXCAD.dir/build.make:346: AppCSXCAD] Error 1
make[4]: *** [CMakeFiles/Makefile2:84: CMakeFiles/AppCSXCAD.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/AppCSXCAD.dir/build.make:86: AppCSXCAD-prefix/src/AppCSXCAD-stamp/AppCSXCAD-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:196: CMakeFiles/AppCSXCAD.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

I tried installing a different version of GDAL to see if that would fix it. Any ideas?

obstinacyvalue, 2022-02-03:
This was addressed in the forums. I will copy that forum chain over and close the ticket.

thliebig, 2022-12-31:
You never did and the forum is not down unfortunately.... Please re-open if this is still an issue

closed 2022-12-31 by thliebig