thliebig/openEMS-Project #24: build failure

evanfoss, 2023-01-23:
I am trying to install openEMS and I have the following log file. 

-- The CXX compiler identification is GNU 10.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CMAKE_INSTALL_PREFIX: /home/user1/opt/openEMS
-- Build AppCSXCAD: YES
-- with MPI: 0
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.ChPxdluuCL
[  2%] Creating directories for 'fparser'
[  5%] No download step for 'fparser'
[  7%] No update step for 'fparser'
[ 10%] No patch step for 'fparser'
[ 12%] Performing configure step for 'fparser'
-- The CXX compiler identification is GNU 10.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:7 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.ChPxdluuCL/fparser-prefix/src/fparser-build
[ 15%] Performing build step for 'fparser'
[ 33%] Building CXX object CMakeFiles/fparser.dir/fparser.cc.o
[ 66%] Building CXX object CMakeFiles/fparser.dir/fpoptimizer.cc.o
[100%] Linking CXX shared library libfparser.so
[100%] Built target fparser
[ 17%] Performing install step for 'fparser'
Consolidate compiler generated dependencies of target fparser
[100%] Built target fparser
Install the project...
-- Install configuration: "Release"
-- Installing: /home/user1/opt/openEMS/lib/libfparser.so.4.5.1
-- Up-to-date: /home/user1/opt/openEMS/lib/libfparser.so.4
-- Up-to-date: /home/user1/opt/openEMS/lib/libfparser.so
-- Installing: /home/user1/opt/openEMS/include/fparser.hh
[ 20%] Completed 'fparser'
[ 20%] Built target fparser
[ 22%] Creating directories for 'CSXCAD'
[ 25%] No download step for 'CSXCAD'
[ 27%] No update step for 'CSXCAD'
[ 30%] No patch step for 'CSXCAD'
[ 32%] Performing configure step for 'CSXCAD'
-- The CXX compiler identification is GNU 10.3.0
-- The C compiler identification is GNU 10.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.31.1") 
-- Found Git repository, CSXCAD version tag: v0.6.2-123-gc29742b
[... 712 lines cut ...]
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/WriteHDF5.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/calcTLPort.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/ReadHDF5Dump.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/ConvertHDF5_VTK.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/plotRefl.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/DFT_time2freq.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/AddStripLinePort.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/FinishQueue.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/Add2Queue.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/AddMSLPort.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/ReadHDF5Attribute.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/AddCPWPort.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/calcWGPort.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/WriteOpenEMS.m
-- Installing: /home/user1/opt/openEMS/share/openEMS/matlab/AddCoaxialPort.m
-- Installing: /home/user1/opt/openEMS/bin/nf2ff
-- Set runtime path of "/home/user1/opt/openEMS/bin/nf2ff" to "/home/user1/opt/openEMS/lib:/usr/lib64/libhdf5.so:/usr/lib64/libz.so:/usr/lib64/libdl.so:/usr/lib64/libm.so"
-- Installing: /home/user1/opt/openEMS/lib/libnf2ff.so.0.1.0
-- Up-to-date: /home/user1/opt/openEMS/lib/libnf2ff.so.0
-- Set runtime path of "/home/user1/opt/openEMS/lib/libnf2ff.so.0.1.0" to "/home/user1/opt/openEMS/lib:/usr/lib64/libhdf5.so:/usr/lib64/libz.so:/usr/lib64/libdl.so:/usr/lib64/libm.so"
-- Up-to-date: /home/user1/opt/openEMS/lib/libnf2ff.so
-- Installing: /home/user1/opt/openEMS/include/openEMS/nf2ff.h
[ 72%] Completed 'openEMS'
[ 72%] Built target openEMS
make: *** [Makefile:91: all] Error 2

thliebig, 2023-01-23:
Well that is a bit weird. It found VTK 9.0.1 and thus should have requested Qt5 (which it found) but it requested Qt4 instead? That does make little sense? see [QCSXCAD CMake](https://github.com/thliebig/QCSXCAD/blob/master/CMakeLists.txt#L118)

This could only mean that VTK_QT_VERSION was set to 4? Which would mean VTK was build with Qt4 support instead of Qt5? What Linux is this and where does VTK come from?

johannDE, 2023-04-03:
I'm having pretty much the same issue, with the following log file
[build_20230403_133636.log](https://github.com/thliebig/openEMS-Project/files/11138308/build_20230403_133636.log)
How can I make it work?

thliebig, 2023-10-10:
Much has changes in the build-system, everything works for me. Please re-open if the problem still exists... and please report which Linux Distro you are using

closed 2023-10-10 by thliebig