thliebig/openEMS-Project #37: python CalcPort

rlarv, 2023-02-21:
Hello,

  I stumbled over this:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "myplace/StripLine2MSL.py", line 154, in <module>
    p.CalcPort( Sim_Path, f, ref_impedance = 50.0)
  File "$HOME/.local/lib/python3.9/site-packages/openEMS-0.0.33-py3.9-linux-x86_64.egg/openEMS/ports.py", line 130, in CalcPort
    self.ut_inc = 0.5 * ( self.ut_tot + self.it_tot * self.Z_ref )
AttributeError: 'MSLPort' object has no attribute 'ut_tot'
```

When invoking CalcPort as
   CalcPort( Sim_Path, f, ref_impedance = 50.0)

Using an integer 50 instead of the float 50.0 fixes the traceback.

The issue seems to be in ports.py somewhere here
````
        if type(self.Z_ref) == float:
            self.ut_inc = 0.5 * ( self.ut_tot + self.it_tot * self.Z_ref )
            self.it_inc = 0.5 * ( self.it_tot + self.ut_tot / self.Z_ref )
            self.ut_ref = self.ut_tot - self.ut_inc
            self.it_ref = self.it_inc - self.it_tot
```

thliebig, 2023-02-21:
The correct repository for this issue would have been openEMS. Not openEMS-Project which is only a "Meta" Project to connect all the dependencies, create documentation and simplify the build process...
But I can see how that is hard to figure out from the outside ;)
I will have a look what is going on here...

thliebig, 2023-02-21:
I don't know how to reproduce the problem. I'm not sure at all why this check for float is there in the first place.
The calculation of the time-domain signals should always be possible?
What tutorial or script did you run?

rlarv, 2023-02-22:
Here is a piece, which I'm currently playing with.

[StripLine2MSL.py.gz](https://github.com/thliebig/openEMS-Project/files/10802492/StripLine2MSL.py.gz)

That is certainly very rough in many aspects, but perhaps helps to reveal what the check for "float" was meant to do.

rlarv, 2023-02-22:
I forgot, yes the actual simulation in time domain does work, the problem is in CalcPort() which is used for plotting.

thliebig, 2023-02-22:
okay I have found and understood it (I think)
The check for float was meant to not calculate incoming voltages and currents in time-domain (TD) in case the reference impedance was frequency dependent (aka an array) 
But the MSL port was never tested with a scalar ref impedance and that it never even calculated any TD voltages.

I have fixed both issues... please give it a try and close if it is fixed for you...

rlarv, 2023-02-22:
Thank You very much, Your commit fixed the issue for me.

closed 2023-02-22 by rlarv