thliebig/openEMS-Project #40: ImportError: DLL load failed while importing CSXCAD

milen-prg, 2023-03-04:
Windows 10 Pro 64 bit, Python 3.11 64 bit, I successful install the packages:

openEMS_x64_v0.0.35-99-gecf0c16_msvc
pip install CSXCAD-0.6.2-cp311-cp311-win_amd64.whl
pip install openEMS-0.0.33-cp311-cp311-win_amd64.whl

BUT, when try any of the examples in:
openEMS_x64_v0.0.35-99-gecf0c16_msvc\openEMS\python\Tutorials

it returns ERRRR:

C:\Users\mlap\Downloads\openEMS_x64_v0.0.35-99-gecf0c16_msvc\openEMS\python\Tutorials>Simple_Patch_Antenna.py
Traceback (most recent call last):
  File "C:\Users\mlap\Downloads\openEMS_x64_v0.0.35-99-gecf0c16_msvc\openEMS\python\Tutorials\Simple_Patch_Antenna.py", line 12, in <module>
    from CSXCAD  import ContinuousStructure
  File "C:\Users\mlap\AppData\Local\Programs\Python\Python311\Lib\site-packages\CSXCAD\__init__.py", line 19, in <module>
    from CSXCAD.CSXCAD import ContinuousStructure
ImportError: DLL load failed while importing CSXCAD: The specified module could not be found.

thliebig, 2023-03-04:
Did you follow the Windows install [instruction](https://docs.openems.de/python/install.html#setup)? Especially to set the environment variables?

milen-prg, 2023-03-05:
I apologize, indeed I did not set the environment variable. Additionally was extra requirement to the h5py package. After that all works!
Thank, you!

closed 2023-03-05 by milen-prg

thliebig, 2023-03-05:
I'm happy to hear that this fixed it for you. Have fun using openEMS ;)