thliebig/openEMS-Project #52: CSXCAD build failure

johannDE, 2023-04-07:
When trying to install openEMS, I get the following error message:

make failed, build incomplete, please see logfile for more details ...
build incomplete, cleaning up tmp dir ...

And this is what the logfile says:

-- The CXX compiler identification is GNU 12.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CMAKE_INSTALL_PREFIX: /home/johann/opt/openEMS
-- Build AppCSXCAD: YES
-- with MPI: 0
-- Configuring done (0.3s)
-- Generating done (0.0s)
-- Build files have been written to: /tmp/tmp.1CL1J2s9Ba
[  2%] Creating directories for 'fparser'
[  5%] No download step for 'fparser'
[  7%] No update step for 'fparser'
[ 10%] No patch step for 'fparser'
[ 12%] Performing configure step for 'fparser'
CMake Warning (dev) at CMakeLists.txt:5 (PROJECT):
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The CXX compiler identification is GNU 12.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:7 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done (0.2s)
-- Generating done (0.0s)
-- Build files have been written to: /tmp/tmp.1CL1J2s9Ba/fparser-prefix/src/fparser-build
[ 15%] Performing build step for 'fparser'
[ 33%] Building CXX object CMakeFiles/fparser.dir/fparser.cc.o
[ 66%] Building CXX object CMakeFiles/fparser.dir/fpoptimizer.cc.o
[100%] Linking CXX shared library libfparser.so
[100%] Built target fparser
[ 17%] Performing install step for 'fparser'
[100%] Built target fparser
Install the project...
-- Install configuration: "Release"
-- Installing: /home/johann/opt/openEMS/lib/libfparser.so.4.5.1
-- Up-to-date: /home/johann/opt/openEMS/lib/libfparser.so.4
-- Up-to-date: /home/johann/opt/openEMS/lib/libfparser.so
-- Up-to-date: /home/johann/opt/openEMS/include/fparser.hh
[ 20%] Completed 'fparser'
[ 20%] Built target fparser
[ 22%] Creating directories for 'CSXCAD'
[ 25%] No download step for 'CSXCAD'
[ 27%] No update step for 'CSXCAD'
[ 30%] No patch step for 'CSXCAD'
[ 32%] Performing configure step for 'CSXCAD'
CMake Warning (dev) at CMakeLists.txt:9 (PROJECT):
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The CXX compiler identification is GNU 12.2.1
-- The C compiler identification is GNU 12.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.40.0") 
-- Found Git repository, CSXCAD version tag: v0.6.2-124-gb5919a6
-- FPARSER_ROOT_DIR: /home/johann/opt/openEMS
-- fparser_INCLUDE_DIR: /home/johann/opt/openEMS/include
-- TinyXML_ROOT_DIR 
-- Found TinyXML: /usr/lib/libtinyxml.so  
-- TinyXML_INCLUDE_DIR: /usr/include
-- TinyXML_LIBRARY: /usr/lib/libtinyxml.so
-- Found HDF5: /usr/lib/libhdf5.so;/usr/lib/libsz.so;/usr/lib/libz.so;/usr/lib/libdl.a;/usr/lib/libm.so (found suitable version "1.14.0", minimum required is "1.8") found components: C HL 
-- Using header-only CGAL
-- Targetting Unix Makefiles
-- Using /usr/bin/c++ compiler.
-- Found GMP: /usr/lib/libgmp.so  
-- Found MPFR: /usr/lib/libmpfr.so  
-- Found Boost: /usr/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found suitable version "1.81.0", minimum required is "1.66")  
-- Boost include dirs: /usr/include
-- Boost libraries:    
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Using gcc version 4 or later. Adding -frounding-math
-- Found Boost: /usr/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found suitable version "1.81.0", minimum required is "1.46") found components: thread system date_time serialization chrono 
-- Found MPI_C: /usr/lib/libmpi.so (found version "3.1") 
-- Found MPI_CXX: /usr/lib/libmpi_cxx.so (found version "3.1") 
-- Found MPI: TRUE (found version "3.1")  
CMake Deprecation Warning at /usr/lib/cmake/vtk/vtk-config.cmake:75 (message):
  The new name for the 'vtkIOGeometry' component is 'IOGeometry'
Call Stack (most recent call first):
  CMakeLists.txt:139 (find_package)

CMake Deprecation Warning at /usr/lib/cmake/vtk/vtk-config.cmake:75 (message):
  The new name for the 'vtkIOPLY' component is 'IOPLY'
Call Stack (most recent call first):
  CMakeLists.txt:139 (find_package)

CMake Warning at /usr/lib/cmake/vtk/VTK-vtk-module-find-packages.cmake:1966 (find_package):
  By not providing "Findnlohmann_json.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "nlohmann_json", but CMake did not find one.

  Could not find a package configuration file provided by "nlohmann_json"
  with any of the following names:

    nlohmann_jsonConfig.cmake
    nlohmann_json-config.cmake

  Add the installation prefix of "nlohmann_json" to CMAKE_PREFIX_PATH or set
  "nlohmann_json_DIR" to a directory containing one of the above files.  If
  "nlohmann_json" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /usr/lib/cmake/vtk/vtk-config.cmake:152 (include)
  CMakeLists.txt:139 (find_package)

-- Found LZMA: /usr/lib/liblzma.so (found version "5.4.2") 
-- Found LZ4: /usr/lib/liblz4.so (found version "1.9.4") 
-- Found double-conversion: /usr/lib/libdouble-conversion.so  
-- Found TIFF: /usr/lib/libtiff.so (found version "4.5.0")  
-- Found ZLIB: /usr/lib/libz.so (found version "1.2.13")  
-- Found PNG: /usr/lib/libpng.so (found version "1.6.39") 
-- Found JPEG: /usr/lib/libjpeg.so (found version "80") 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find utf8cpp (missing: utf8cpp_INCLUDE_DIR) 
-- Found package VTK. Using version 9.2.6
CMake Deprecation Warning at /usr/lib/cmake/vtk/vtk-use-file-deprecated.cmake:1 (message):
  The `VTK_USE_FILE` is no longer used starting with 8.90.
Call Stack (most recent call first):
  CMakeLists.txt:142 (include)

-- vtk libraries VTK::IOGeometryVTK::IOPLY
-- Configuring done (2.3s)
CMake Error at /usr/lib/cmake/vtk/VTK-targets.cmake:555 (set_target_properties):
  The link interface of target "VTK::nlohmannjson" contains:

    nlohmann_json::nlohmann_json

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /usr/lib/cmake/vtk/vtk-config.cmake:138 (include)
  CMakeLists.txt:139 (find_package)

-- Generating done (0.0s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
make[2]: *** [CMakeFiles/CSXCAD.dir/build.make:93: CSXCAD-prefix/src/CSXCAD-stamp/CSXCAD-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:117: CMakeFiles/CSXCAD.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

I've been trying to install it for a while now as well as looking into the CmakeLists.txt file and trying to modify it, but to no avail. I really need this to work,what can I do?

thliebig, 2023-04-07:
You miss the most important information... what Linux are you using?

johannDE, 2023-04-07:
Sorry, I use Manjaro.

thliebig, 2023-04-07:
Well that is weird, it works for me fine, I see this line:
`-- Found nlohmann_json: /usr/share/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found version "3.11.2")`

Maybe you need to install "nlohmann-json" ? I'm not even sure what that is and why vtk 9.2 now needs this... vtk's dependencies are a nightmare...

johannDE, 2023-04-07:
Sorry the "close with comment" is a bit confusing to me, it's my first issue.

I'll try looking for the file in my system then.

johannDE, 2023-04-07:
I don't have  /nlohmann_json folder it seems. Could you show me how to fix that?

thliebig, 2023-04-07:
Did you install the package I mentioned?
`sudo pamac install nlohmann-json`

johannDE, 2023-04-07:
I just did, build successful. You really saved me here, I need this for an exam next week. Thank you very much!

thliebig, 2023-04-07:
You need openEMS for an exam? Glad I could help.. (Now the Close with comment would have been correct ;) )

closed 2023-04-07 by thliebig

johannDE, 2023-04-07:
Since I don't want to use a virtual machine for matlab, I use openEMS as a antenna simulator. Next week's exam will mostly consist of antenna simualtions. (gotcha, I'll make sure to remember that :-) )