thliebig/openEMS-Project #66: Like to run OpenEMS on a patch antenna under Windows.

TahhaR, 2023-05-12:
Hello people,
i created my own patch antenna and want to simulate it through OpenEMS but i get problems during or after a long while of simulation ( the last one lasted for more than one hour of copilation). I generate an automatic script with Freecad_to_OpenEMS_Export. I run the already existing examples and they ran easily and the Energy was not as the same as mine . So if you could help it would be nice. Thanks in advance.
[image]

LubomirJagos, 2023-05-12:
Hi, really depends on your script what you generated, since you use that GUI addon your simulation should be generated right in sense that it displayed in AppCSXCAD right (all parts were placed on correct place). Most common issue since energy is not decreasing could be that mesh is generated wrong, must be corssing PEC material, substrate, also maybe some of generated gridlines are too close to each other therefore timestep is too small.
It would be more helpful to post at least screenshot of your antenna in AppCSXCAD or generated .m script

LubomirJagos, 2023-05-12:
Looking that you opened this as issue is not right, you can post it into discussion, but this is not openEMS issue, openEMS is running fine, but your simulation is generated wrong, so if you can please close it here and create discussion topic for this.

TahhaR, 2023-05-12:
Ok Thanks for the answer. i ll open a new discussion topic.

closed 2023-05-12 by TahhaR

TahhaR, 2023-05-12:
Hi,

I started the discussion if you want more infos don’t hesitate to ask o
write me anytime.
It’s the first time i start a discussion on Github 😆

TahhaR, 2023-05-20:
Hi Lubomir,
can you please help me out to understand the Meshing, because i doubt on it
and i have red that´s very important for OpenEMS. That´s why, i would do it
manually. If you do have some examples please let me know. Thank you in
advance and have a nice weekend.
Best regards
Tahhar from Github