thliebig/openEMS-Project #90: OpenEMS build failure on Ubuntu 20.04

qiank10, 2023-06-17:
I am trying to build openEMS and encountering some errors with CAGL. The build log is as follows: 

-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CMAKE_INSTALL_PREFIX: /home/kunqian/opt/openEMS
-- Build AppCSXCAD: YES
-- with MPI: 0
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.yOQQ35sfXI
Scanning dependencies of target fparser
[  2%] Creating directories for 'fparser'
[  5%] No download step for 'fparser'
[  7%] No patch step for 'fparser'
[ 10%] No update step for 'fparser'
[ 12%] Performing configure step for 'fparser'
-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.yOQQ35sfXI/fparser-prefix/src/fparser-build
[ 15%] Performing build step for 'fparser'
Scanning dependencies of target fparser
[ 33%] Building CXX object CMakeFiles/fparser.dir/fparser.cc.o
[ 66%] Building CXX object CMakeFiles/fparser.dir/fpoptimizer.cc.o
[100%] Linking CXX shared library libfparser.so
[100%] Built target fparser
[ 17%] Performing install step for 'fparser'
[100%] Built target fparser
Install the project...
-- Install configuration: "Release"
-- Installing: /home/kunqian/opt/openEMS/lib/libfparser.so.4.5.1
-- Up-to-date: /home/kunqian/opt/openEMS/lib/libfparser.so.4
-- Up-to-date: /home/kunqian/opt/openEMS/lib/libfparser.so
-- Up-to-date: /home/kunqian/opt/openEMS/include/fparser.hh
[ 20%] Completed 'fparser'
[ 20%] Built target fparser
Scanning dependencies of target CSXCAD
[ 22%] Creating directories for 'CSXCAD'
[ 25%] No download step for 'CSXCAD'
[ 27%] No patch step for 'CSXCAD'
[ 30%] No update step for 'CSXCAD'
[ 32%] Performing configure step for 'CSXCAD'
-- Found Git repository, CSXCAD version tag: v0.6.2-124-gb5919a6
-- FPARSER_ROOT_DIR: /home/kunqian/opt/openEMS
-- fparser_INCLUDE_DIR: /home/kunqian/opt/openEMS/include
-- TinyXML_ROOT_DIR 
-- TinyXML_INCLUDE_DIR: /usr/include
-- TinyXML_LIBRARY: /usr/lib/x86_64-linux-gnu/libtinyxml.so
-- HDF5: Using hdf5 compiler wrapper to determine C configuration
-- Using header-only CGAL
-- Targetting Unix Makefiles
-- Using /bin/c++ compiler.
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.48")  
-- Boost include dirs: /include
-- Boost libraries:    
-- Using gcc version 4 or later. Adding -frounding-math
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.46") found components: thread system date_time serialization chrono 
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- Found package VTK. Using version 7.1.1
-- vtk libraries vtkIOGeometryvtkCommonCorevtksysvtkCommonDataModelvtkCommonMathvtkCommonMiscvtkCommonSystemvtkCommonTransformsvtkCommonExecutionModelvtkIOCore/usr/lib/x86_64-linux-gnu/libz.sovtkIOLegacyvtkIOPLY
-- Configuring done
CMake Error in src/CMakeLists.txt:
  Imported target "CGAL" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Warning at src/CMakeLists.txt:86 (add_library):
  Cannot generate a safe runtime search path for target CSXCAD because files
  in some directories may conflict with libraries in implicit directories:

    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/kunqian/anaconda3/lib
    runtime library [libmpfr.so.6] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/kunqian/anaconda3/lib
    runtime library [libgmp.so.10] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/kunqian/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Error in src/CMakeLists.txt:
  Imported target "CGAL" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
make[2]: *** [CMakeFiles/CSXCAD.dir/build.make:108: CSXCAD-prefix/src/CSXCAD-stamp/CSXCAD-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:138: CMakeFiles/CSXCAD.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

thliebig, 2023-10-10:
Much has changes with the build-system, can you confirm this problem still exists? I must say I do not test much on this old/legacy systems

oberstet, 2023-11-17:
fwiw, just some comments from the sideline:

I wouldn't mix building partially with system libraries, and libraries coming bundled with Anaconda. in fact I wouldn't build under Anaconda at all, but build everything from plain vanilla upstream Python sources includes Python itself. I would also upgrade your system to the latest LTS, which works flawlessly these days.

building works without problems on Ubuntu 22.04 LTS on vanilla upstream CPython 3.11 with latest openEMS master. though I would bet it's your Anaconda / mixing stuff at fault ..

I have not yet tried building on other Pythons. I'd be interested in [PyPy](https://www.pypy.org/) for improving speed. there might be some things then to fix, I've also noted some other minor details which could be improved, and proper PyPI packaging would be nice too! anyways, different things, just my 2cts;)

closed 2024-04-01 by thliebig