thliebig/openEMS-Project #124: ISSUE: python AddMSLPort different from octave, not adding Resistor lumped part

LubomirJagos, 2023-10-03:
Hi,
I'm writing GUI addon for FreeCAD so I have GUI where you define parameters for openEMS simulation and you can generate matlab/octave or python script (now writing python generator) and I was implementing microstrip port and hit this problem:
    - I can generate microstrip port using octave where there is MSL input and output port and for control lumped port all of them with reisstance defined
    - when generating python code lumped port is generated with resistance lumped part
    - MSL port has no resistor lumped part

here is octave code
```matlab
% OpenEMS FDTD Analysis Automation Script
%
% To be run with GNU Octave or MATLAB.
% FreeCAD to OpenEMS plugin by Lubomir Jagos, 
% see https://github.com/LubomirJagos/FreeCAD-OpenEMS-Export
%
% This file has been automatically generated. Manual changes may be overwritten.
%

close all
clear
clc

%% Change the current folder to the folder of this m-file.
if(~isdeployed)
  mfile_name          = mfilename('fullpath');
  [pathstr,name,ext]  = fileparts(mfile_name);
  cd(pathstr);
end

%% constants
physical_constants;
unit    = 0.001; % Model coordinates and lengths will be specified in mm.
fc_unit = 0.001; % STL files are exported in FreeCAD standard units (mm).

%% switches & options
postprocessing_only = 1;
draw_3d_pattern = 0; % this may take a while...
use_pml = 0;         % use pml boundaries instead of mur

currDir = strrep(pwd(), '\', '\\');
display(currDir);

% --no-simulation : dry run to view geometry, validate settings, no FDTD computations
% --debug-PEC     : generated PEC skeleton (use ParaView to inspect)
openEMS_opts = '--no-simulation';

%% prepare simulation folder
Sim_Path = 'simulation_output';
Sim_CSX = 'Mircostrip_Port_1.xml';
[status, message, messageid] = rmdir( Sim_Path, 's' ); % clear previous directory
[status, message, messageid] = mkdir( Sim_Path ); % create empty simulation folder

%% setup FDTD parameter & excitation function
max_timesteps = 1000000;
min_decrement = 1e-05; % 10*log10(min_decrement) dB  (i.e. 1E-5 means -50 dB)
FDTD = InitFDTD( 'NrTS', max_timesteps, 'EndCriteria', min_decrement );

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
% BOUNDARY CONDITIONS
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
BC = {"PEC","PEC","PEC","PEC","PEC","PEC"};
FDTD = SetBoundaryCond( FDTD, BC );

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
% COORDINATE SYSTEM
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
CSX = InitCSX('CoordSystem', 0); % Cartesian coordinate system.
mesh.x = []; % mesh variable initialization (Note: x y z implies type Cartesian).
mesh.y = [];
mesh.z = [];
CSX = DefineRectGrid(CSX, unit, mesh); % First call with empty mesh to set deltaUnit attribute.
[... 279 lines cut ...]
portR = 89.0
portUnits = 1
portDirection = 'z'
port[3] = FDTD.AddLumpedPort(port_nr=3, R=portR*portUnits, start=portStart, stop=portStop, p_dir=portDirection, priority=10000, excite=0)

#######################################################################################################################################
# RUN
#######################################################################################################################################
### Run the simulation
CSX_file = os.path.join(Sim_Path, Sim_CSX)
if not os.path.exists(Sim_Path):
	os.mkdir(Sim_Path)
CSX.Write2XML(CSX_file)
from CSXCAD import AppCSXCAD_BIN
os.system(AppCSXCAD_BIN + ' "{}"'.format(CSX_file))

if not postprocessing_only:
	FDTD.Run(Sim_Path, verbose=3, cleanup=True)
```

this is octave generated model which is RIGHT
[image]

here is python generated model which is WRONG
[image]

thliebig, 2023-10-03:
Looking at the port definition I think the problem is that for the python interface the keyword is "Feed_R" not just "R"....
See [here](https://github.com/thliebig/openEMS/blob/master/python/openEMS/ports.py#L227)

LubomirJagos, 2023-10-03:
yes, that's it, this is missing in python interface documentation for ports here https://docs.openems.de/python/openEMS/ports.html

closed 2023-10-03 by LubomirJagos