thliebig/openEMS-Project #193: The problem of calculating the S-parameters of Magic tee and a rectangular waveguide

Sheptor, 2024-03-09:
# The problem of calculating the S-parameters of Magic tee and a rectangular waveguide
## Introduction

Hello! I am engaged in research on the propagation of an electromagnetic wave in a segment of the magic tee waveguide with cross section 23x10mm^2. But in the simulation results, the sum of the squares of the s-parameters is not equal to 1 (figure 2). I tried to create a waveguide in different ways:
1. Import from `stl` file;
2. Creating waveguide metal walls using AddBox;
3. Filling the simulation space with metal and adding a waveguide cavity using AddBox with epsilon=1 (figure 1). 

### Figure 1. Magic tee (metall hidden)
[image]

### Figure 2. Magic tee (s-parameters)
[image]

# Problem
To verify the results, a rectangular waveguide with a size of 23x10mm^2 was modeled. In the first case: the boundaries of the grid coincide with the section of the waveguide. in the second case, the border is shifted and metal is added to the place of free space (figures 5 - 8).

## Question:
### How does adding metal walls differ from setting boundary conditions ["PEC", "PEC", "PEC", "PEC", "PML_8", "PML_8"]?

### Figгre 3. Empty waveguide
[image]

### Figure 4. Empty waveguide (s-parameters)
[image]

### Figure 5. Waveguide with thin metal walls (metal walls hidden)
[image]

### Figure 6. Waveguide with thin metal walls (metal walls hidden) (s-parameters)
[image]

### Figure 7. The walls of the waveguide are larger than the cavity of the waveguide
[image]

### Figure 8. The walls of the waveguide are larger than the cavity of the waveguide (s-parameters)
[image]

# Code
The simulation was carried out using both the python interface and the octave interface:
https://github.com/Sheptor/test-issue

thliebig, 2024-03-11:
I have not yet had the time to look into your problem in more detail, but from the looks this would have been better suited for the Discussion section and not as an issue? 
I mean it is always possible that an actual issue may be found causing your questions, but usually this is more a "how to use openEMS"  question?

Sheptor, 2024-03-12:
I agree that this applies to the discussion section, but due to inattention, this problem has been posted in this section. A few hours later, I noticed this and duplicated it in the discussion section https://github.com/thliebig/openEMS-Project/discussions/194.
I conducted additional studies with an smoothed grid, the results of which I added to the discussion.

closed 2024-04-01 by thliebig