thliebig/openEMS-Project #197: Homebrew build fails on OS X 14.4.1 (Intel)

horvathp, 2024-03-30:
I'm trying to install openEMS using Homebrew on OS X 14.4.1, Intel i7 CPU, following the installation guide. The install gets stuck. I completely wiped the Homebrew installation and tried again, to the same result. 

Is there a fix for this?

`brew tap thliebig/openems https://github.com/thliebig/openEMS-Project.git
brew install --HEAD openems`

[...]
`==> Installing openems from thliebig/openems
==> cmake .
==> make
Last 15 lines from /Users/hp/Library/Logs/Homebrew/openems/02.make:
-- TinyXML_LIBRARY: /usr/local/lib/libtinyxml.dylib
-- Unable to determine HDF5 C flags from HDF5 wrapper.
CMake Error at /usr/local/Cellar/cmake/3.29.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find HDF5 (missing: HDF5_LIBRARIES HDF5_HL_LIBRARIES) (found
  suitable version "1.14.3", minimum required is "1.8")
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.29.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.29.0/share/cmake/Modules/FindHDF5.cmake:1025 (find_package_handle_standard_args)
  CMakeLists.txt:103 (find_package)

-- Configuring incomplete, errors occurred!
make[2]: *** [CSXCAD-prefix/src/CSXCAD-stamp/CSXCAD-configure] Error 1
make[1]: *** [CMakeFiles/CSXCAD.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so to (not Homebrew/brew or Homebrew/homebrew-core):
  thliebig/openems

/usr/local/Homebrew/Library/Homebrew/utils/github/api.rb:378:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)
	from /usr/local/Homebrew/Library/Homebrew/utils/github/api.rb:272:in `open_rest'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:178:in `search'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:182:in `search_results_items'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:36:in `search_issues'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:62:in `issues_for_formula'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:482:in `fetch_issues'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:477:in `issues'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:533:in `dump'
	from /usr/local/Homebrew/Library/Homebrew/brew.rb:149:in `rescue in <main>'
	from /usr/local/Homebrew/Library/Homebrew/brew.rb:137:in `<main>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:2833:in `block in system': Failed executing: make (BuildError)
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:2769:in `open'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:2769:in `system'
	from /usr/local/Homebrew/Library/Taps/thliebig/homebrew-openems/openems.rb:35:in `install'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:179:in `block (3 levels) in install'
	from /usr/local/Homebrew/Library/Homebrew/extend/kernel.rb:495:in `with_env'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:139:in `block (2 levels) in install'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:1418:in `block in brew'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:3001:in `block (2 levels) in stage'
	from /usr/local/Homebrew/Library/Homebrew/extend/kernel.rb:495:in `with_env'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:3000:in `block in stage'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:129:in `block (2 levels) in unpack'
	from /usr/local/Homebrew/Library/Homebrew/download_strategy.rb:121:in `chdir'
	from /usr/local/Homebrew/Library/Homebrew/download_strategy.rb:113:in `stage'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:125:in `block in unpack'
	from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:75:in `block in run'
	from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:75:in `chdir'
	from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:75:in `run'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:226:in `stage_resource'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:124:in `unpack'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:98:in `stage'
	from /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/3.1.4/lib/ruby/3.1.0/forwardable.rb:238:in `stage'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:2980:in `stage'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:1411:in `brew'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:133:in `block in install'
	from /usr/local/Homebrew/Library/Homebrew/extend/kernel.rb:495:in `with_env'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:125:in `install'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:231:in `<main>'`

horvathp, 2024-03-30:
Pinning HDF5 to 1.10 (and also installing python-setuptools) seems to work.

aWZHY0yQH81uOYvH, 2024-08-15:
This is apparently a bug in CMake: https://gitlab.kitware.com/cmake/cmake/-/issues/25358
and has been [documented in this project](https://docs.openems.de/install/manual.html#troubleshooting) to happen on FreeBSD, but seems to be happening on macOS now too.

Removing the HDF5 1.8 version requirement from the CMakeLists.txt in openEMS, CSXCAD, and AppCSXCAD allows the build to continue with HDF5 version 1.14.3. If the minimum version number is left in there, it only works if it matches the currently installed version of HDF5: `find_package(HDF5 1.14 COMPONENTS C HL REQUIRED)`

Either we should wait for the CMake bug to be fixed, or remove the version requirement for HDF5. I doubt many people have an HDF5 version older than 1.8 nowadays given that was released 17 years ago, but regardless that seems suboptimal.

PR #228 (2024-09-16): Workaround for CMake not finding HDF5
Fixes #197 

On macOS (and also FreeBSD according to the [docs here](https://docs.openems.de/install/manual.html#troubleshooting)), CMake can't find HDF5 when a minimum version number is specified. It's been reported on the CMake bug tracker: https://gitlab.kitware.com/cmake/cmake/-/issues/25358

But for now it's breaking the Homebrew build, so I added a patch that will remove the 1.8 minimum version dependency from the CMakeLists.txt files before building. Homebrew installs the latest version of HDF5, and version 1.8 is 17 years old, so I don't imagine th [... cut]