thliebig/openEMS-Project #198: ImportError : DLL load failed, after installing as per install instructions

MaximGi, 2024-04-01:
Following instructions at [link](https://docs.openems.de/python/install.html#windows) :

- I downloaded v0.0.36 .zip release at [](https://github.com/thliebig/openEMS-Project/releases)
- Extracted at C:\openEMS
- set up global environment variable with `setx OPENEMS_INSTALL_PATH C:\opt\openEMS`
- Installed pre-built modules with (in this order) in a 3.11.5 virtual environment. :
`pip install .\CSXCAD-0.6.3-cp311-cp311-win_amd64.whl`
'openEMS-0.0.36-cp311-cp311-win_amd64.whl'

The following exceptions are raised while trying to import either of both CSXCAD and OpenEMS Modules : 

Open EMS traceback
```import openEMS
Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2023.1.1\plugins\python-ce\helpers\pydev\pydevconsole.py", line 364, in runcode
    coro = func()
           ^^^^^^
  File "<input>", line 1, in <module>
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2023.1.1\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\maxim\python\ml2023\Lib\site-packages\openEMS\__init__.py", line 7, in <module>
    import CSXCAD
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2023.1.1\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\maxim\python\ml2023\Lib\site-packages\CSXCAD\__init__.py", line 19, in <module>
    from CSXCAD.CSXCAD import ContinuousStructure
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2023.1.1\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: DLL load failed while importing CSXCAD: Le module spécifié est introuvable.
```

CSXCAD traceback
```import CSXCAD
Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2023.1.1\plugins\python-ce\helpers\pydev\pydevconsole.py", line 364, in runcode
    coro = func()
           ^^^^^^
  File "<input>", line 1, in <module>
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2023.1.1\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\maxim\python\ml2023\Lib\site-packages\CSXCAD\__init__.py", line 19, in <module>
    from CSXCAD.CSXCAD import ContinuousStructure
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2023.1.1\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: DLL load failed while importing CSXCAD: Le module spécifié est introuvable.
```

thliebig, 2024-04-01:
Can you please try this with a plain simple python (not via JetBrains)? It seems to me that JetBrains uses its own wired import method (not sure)?
What I prefer on Windows is [WinPython](https://winpython.github.io/) and this is the only way I have tested this so far.

thliebig, 2024-04-01:
What you can try too is in your script tell python where to look for the *.dll with:

`os.add_dll_directory(r 'C:\opt\openEMS')`

MaximGi, 2024-04-01:
OK, I found a way to keep working with Pycharm + virtual env.

It is possible to use this workaround :
> What you can try too is in your script tell python where to look for the *.dll with:
> 
> `os.add_dll_directory(r'C:\opt\openEMS')`

Or, just add C:\opt\openEMS or whatever the directory is as as source directory in the pycharm project.

In a more general way, do you think it is possible to package all of this into a Pypi namespace so openEMS can be installed directly with pip ? I'm only suggesting, not demanding because I'm not competent enough to do it.

closed 2024-04-01 by MaximGi