thliebig/openEMS-Project #203: Installation on Windows 11

jones-jp, 2024-04-24:
Hi,

I followed the instructions here: https://docs.openems.de/install.html (Basically just extracting the downloaded zip folder into C drive).

I did the next steps give here: https://docs.openems.de/python/install.html#windows
git clone --recursive https://github.com/thliebig/openEMS-Project.git
cd openEMS-Project/CSXCAD/python

So far so good...

But when attempting to install with:
python setup.py build_ext -IC:\opt\openEMS\include -LC:\opt\openEMS

I get the following errors (copying here only the last few lines, but the same error is repeated 100 times):
C:\OpenSourceSI\openEMS-Project\CSXCAD\python\CSXCAD\CSProperties.cpp(47892): error C2039: 'LEtype': is not a member of 'CSPropLumpedElement'
C:\OpenSourceSI\openEMS\include\CSXCAD/CSPropLumpedElement.h(26): note: see declaration of 'CSPropLumpedElement'
C:\OpenSourceSI\openEMS-Project\CSXCAD\python\CSXCAD\CSProperties.cpp(47893): error C2039: 'LEtype': is not a member of 'CSPropLumpedElement'
C:\OpenSourceSI\openEMS\include\CSXCAD/CSPropLumpedElement.h(26): note: see declaration of 'CSPropLumpedElement'
C:\OpenSourceSI\openEMS-Project\CSXCAD\python\CSXCAD\CSProperties.cpp(47893): error C2039: 'LEtype': is not a member of 'CSPropLumpedElement'
C:\OpenSourceSI\openEMS\include\CSXCAD/CSPropLumpedElement.h(26): note: see declaration of 'CSPropLumpedElement'
C:\OpenSourceSI\openEMS-Project\CSXCAD\python\CSXCAD\CSProperties.cpp(47893): error C2039: 'LEtype': is not a member of 'CSPropLumpedElement'
C:\OpenSourceSI\openEMS\include\CSXCAD/CSPropLumpedElement.h(26): note: see declaration of 'CSPropLumpedElement'
C:\OpenSourceSI\openEMS-Project\CSXCAD\python\CSXCAD\CSProperties.cpp(47898): error C2039: 'LEtype': is not a member of 'CSPropLumpedElement'
C:\OpenSourceSI\openEMS\include\CSXCAD/CSPropLumpedElement.h(26): note: see declaration of 'CSPropLumpedElement'
C:\OpenSourceSI\openEMS-Project\CSXCAD\python\CSXCAD\CSProperties.cpp(47898): fatal error C1003: error count exceeds 100; stopping compilation
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.39.33519\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

Please help with fixing this issue.

Regards,
Jones

LubomirJagos42, 2024-04-24:
this is not solution, let's say workaround for now to exclude inductance extension which you will not use for python interface for now:
```
# downgrade openEMS to version without new stuff lumped inductance so it's possible to compile
cd ~/Downloads/openEMS-Project/openEMS
git reset --hard 5f36e7f3a2367123f00999491a069aed50c6f244
```

thliebig, 2024-04-24:
The zip should come with pre-build python packages. Have a look in the python folder and the readme on how to just install these packages...

jones-jp, 2024-04-30:
Hi LubomirJagos,
Thanks for your response! I was trying to install in Windows in my work PC, I was looking for an open source alternative to tools like Hyperlynx. For now, I have given up my efforts to get this working in Windows, but in my Linux system at home now I am trying to get this work.

Hi thliebig,
Thanks for your suggestion! I had tried that too, but it seems like it doesn't work on the latest version of Python. I was trying on my work PC and every time I try to install something, I'll need to get admin rights from the IT department and I gave up trying on this. But I'm trying to get this work on my personal Linux system.

Really happy to see your replies. Sorry I took a while to get back to you.

Jones

karinemariew, 2024-10-20:
> this is not solution, let's say workaround for now to exclude inductance extension which you will not use for python interface for now:
>
> ```
> # downgrade openEMS to version without new stuff lumped [...]

@LubomirJagos42 I attempted this workaround and it still throws the error ('LEtype': is not a member of 'CSPropLumpedElement') when I attempt the following command:

 python setup.py build_ext -I "C:\opt\openEMS\include" -L "C:\opt\openEMS"

thliebig, 2024-10-22:
I think you are trying to build the newest version of the python extension against a not up to date openEMS/CSXCAD library (dll). That will no work.. Please only download the pre-build version and just install the pre-compiled python modules or at least use the  original (older) python sources...

thliebig, 2024-10-22:
Building software on Windows is a major pain the in the ...  I really think you should switch to a proper operating system :smirk:

SupaTuba, 2024-10-22:
@thliebig Lol. No worries; my next step was going to be to attempt a Linux install. 

I will try the Pre-Build version again, in case I messed it up the first time.