thliebig/openEMS-Project #216: License ??

the-moog, 2024-06-24:
I looking at the current state of  FOSS EM simulation solutions I note your project does not list a license.

Though it is in no way mandatory to decide upon a license for your code, you may want to read the GitHub/Microsoft page on licensing within it's terms of service as it effects your future rights to your own intellectual property and that of those who derive IP from it.

See
[https://docs.github.com/en/site-policy/github-terms/github-terms-of-service](https://docs.github.com/en/site-policy/github-terms/github-terms-of-service)
[https://github.com/readme/guides/open-source-licensing](https://github.com/readme/guides/open-source-licensing)

This SE entry summaraises the issue:
[What can I assume if a publicly published project has no license](https://opensource.stackexchange.com/questions/1720/what-can-i-assume-if-a-publicly-published-project-has-no-license)

And this is the GitHub default
[https://choosealicense.com/no-permission/](https://choosealicense.com/no-permission/)

Note the carefully worded '...within GitHub...' includes ChatGPT, which can and will offer portions of your own IP to others as coding suggestions, effectively giving away your IP under a sort of easy read obfustiction.

--

Slightly off-topic. I am engineer, not a lawyer though I do take an interest in FOSS. I will include this here as I find it an interesting topic. I hope others will too and have their own opinions on the subject.  To me it seems software IP owrnership is one of those 'foot shuffling' things that everybody is hoping will just go away....
Though still a legal grey area, conventional wisdom considers software source code intellectional property as "written works" under the [Berne Convention](https://en.wikipedia.org/wiki/Berne_Convention).  This needs urgent review as times have changed since even this clarrification was made.

1. The Berne Convention, 1886 (not a typo) created a default set of rights which expire after 50 years. Which is daft when it comes to software which may have a useful lifecycle measured in months. [ref](https://en.wikipedia.org/wiki/Abandonware)
2. Britian, then the US, then most other contries adopted it as a foundation of intellectual property law in the late 1980's
3. Now, in the 2020's Abandonware and IP theft both are rather pressing issues
4. ChatGPT was science fiction and GitHub did not exist when it became law
5. The phrase 'Git' was still a British English insult when it was last updated in 1971 under the [Paris Act](https://www.wipo.int/wipolex/en/treaties/textdetails/12800).
6. The concept of distributed SCM did not exist as [Linux Torvolds](https://en.wikipedia.org/wiki/Linus_Torvalds), who originally [wrote git/scm](https://en.wikipedia.org/wiki/Git), was a toddler when the Berne Convention on this topic last met in 1971.
7. I doubt Linus (or indeed anybody) saw Git being the foundation of a number of global, commercially ownded code repositories.
8. Linux created both the Linux project in the early 1990's and the Git project itself in the mid 2000's, both as a response to his frustrations with commercial IP protection of source code.

thomaslepoix, 2024-06-25:
The project is split across various other repositories. This repository is mostly a convenience for dependency management & build system stuff (and also seems to host some documentation pages now). The actual project is under GPL3 / LGPL3.

cristidbr, 2024-11-30:
Myself sorry to say it, but licensing information is **important** - and that's currently missing in the openEMS GitHub project. I can help with this one and do a PR that fixes the unclear licensing.

🚫 Blocker: is GPL3 / LGPLv3 what this project needs? Do all maintainers support this licensing model?

@thomaslepoix splitting across repositories is not unheard of, and SPDX made some progress in standardising the software Bill-Of-Materials through ISO/IEC 5962:2021.

https://spdx.dev/about/overview/

 TLDR: Please do a LICENSE.md, me currently looking at issues to figure out how to get this software installing

thliebig, 2024-12-01:
I would not be sure what license this "meta" project should have or why this would be an issue since this does not even really contain any software...

closed 2024-12-01 by thliebig