thliebig/openEMS-Project #218: Error in installation

nuclearrambo, 2024-06-29:
While running the install script I am coming across these errors. Can anyone help me get past these so I can successfully install OpenEMS?

`-- Build files have been written to: /tmp/tmp.YLCiiFlJam/AppCSXCAD-prefix/src/AppCSXCAD-build
[ 90%] Performing build step for 'AppCSXCAD'
[ 20%] Automatic MOC for target AppCSXCAD
[ 97%] Linking CXX executable nf2ff
[ 97%] Built target openEMS
[ 98%] Building CXX object CMakeFiles/openEMS_bin.dir/main.cpp.o
[ 20%] Built target AppCSXCAD_autogen
[ 60%] Building CXX object CMakeFiles/AppCSXCAD.dir/main.cpp.o
[ 60%] Building CXX object CMakeFiles/AppCSXCAD.dir/AppCSXCAD_autogen/mocs_compilation.cpp.o
[ 80%] Building CXX object CMakeFiles/AppCSXCAD.dir/AppCSXCAD.cpp.o
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_global_init@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_easy_perform@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_slist_free_all@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_easy_setopt@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_easy_init@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_slist_append@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_easy_cleanup@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_global_cleanup@CURL_OPENSSL_4'
collect2: error: ld returned 1 exit status
make[5]: *** [nf2ff/CMakeFiles/nf2ff_bin.dir/build.make:105: nf2ff/nf2ff] Error 1
make[4]: *** [CMakeFiles/Makefile2:254: nf2ff/CMakeFiles/nf2ff_bin.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
[100%] Linking CXX executable openEMS
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_global_cleanup@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_global_init@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_easy_perform@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_easy_init@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_slist_free_all@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_easy_setopt@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_slist_append@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so: undefined reference to `curl_easy_cleanup@CURL_OPENSSL_4'
collect2: error: ld returned 1 exit status
make[5]: *** [CMakeFiles/openEMS_bin.dir/build.make:125: openEMS] Error 1
make[4]: *** [CMakeFiles/Makefile2:202: CMakeFiles/openEMS_bin.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/openEMS.dir/build.make:86: openEMS-prefix/src/openEMS-stamp/openEMS-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:144: CMakeFiles/openEMS.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX executable AppCSXCAD
[100%] Built target AppCSXCAD
[ 92%] Performing install step for 'AppCSXCAD'
[ 20%] Automatic MOC for target AppCSXCAD
[ 20%] Built target AppCSXCAD_autogen
[100%] Built target AppCSXCAD
Install the project...
-- Install configuration: "Release"
-- Installing: /home/nuclearrambo/opt/openEMS/bin/AppCSXCAD
-- Set runtime path of "/home/nuclearrambo/opt/openEMS/bin/AppCSXCAD" to "/home/nuclearrambo/opt/openEMS/lib:/usr/lib/x86_64-linux-gnu/hdf5/serial"
-- Up-to-date: /home/nuclearrambo/opt/openEMS/bin/AppCSXCAD.sh
[ 95%] Completed 'AppCSXCAD'
[ 95%] Built target AppCSXCAD
make: *** [Makefile:91: all] Error 2
`

jabowery, 2024-10-30:
Lots of people have this problem.

g3grau, 2024-12-18:
I had a bunch of related problems.
I initially tried it on openSuse Tumbleweed. I had an older VTK installed, it failed, then I retried it with the recent one 9.4.0 with no luck (VTK build was ok).

Then I came across [IIC-OSIC-TOOLS](https://github.com/iic-jku/IIC-OSIC-TOOLS) which is an Ubuntu20 container with many tools preinstalled - unfortunately openEMS without the python extension.
Trying to rebuild openEMS with the VTK7.1.1 in the container worked fine but executing a python script resulted in strange errors related to cython and VTK. Trying to rebuild VTK7 from source failed, same for VTK8. VTK9.4 installation in Ubuntu succeeded, but building openEMS failed. So I have a solution that compiled but doesn't run and another one that doesn't compile :-D Running out of ideas now. 
I had a brief look at the Makefile2 (which is called on the Makefile:91) but I'm clueless how to dig deeper, Makefiles are not for me.

```
# openSuse Tumbleweed: 
./update_openEMS.sh ~/opt/openEMS --with-hyp2mat --with-CTB --python
-- The CXX compiler identification is GNU 14.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CMAKE_INSTALL_PREFIX: /root/opt/openEMS
-- Build AppCSXCAD: YES
-- with MPI: 0
-- Configuring done (1.0s)
-- Generating done (0.0s)
-- Build files have been written to: /tmp/tmp.wGyYLlKJA6
[  2%] Creating directories for 'fparser'
[  5%] No download step for 'fparser'
[  7%] No update step for 'fparser'
[ 10%] No patch step for 'fparser'
[ 12%] Performing configure step for 'fparser'
CMake Warning (dev) at CMakeLists.txt:5 (PROJECT):
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The CXX compiler identification is GNU 14.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:7 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- Configuring done (0.6s)
-- Generating done (0.0s)
-- Build files have been written to: /tmp/tmp.wGyYLlKJA6/fparser-prefix/src/fparser-build
[ 15%] Performing build step for 'fparser'
[ 66%] Building CXX object CMakeFiles/fparser.dir/fpoptimizer.cc.o
[ 66%] Building CXX object CMakeFiles/fparser.dir/fparser.cc.o
[100%] Linking CXX shared library libfparser.so
[100%] Built target fparser
[ 17%] Performing install step for 'fparser'
[100%] Built target fparser
Install the project...
-- Install configuration: "Release"
-- Installing: /root/opt/openEMS/lib/libfparser.so.4.5.1
-- Installing: /root/opt/openEMS/lib/libfparser.so.4
-- Installing: /root/opt/openEMS/lib/libfparser.so
-- Installing: /root/opt/openEMS/include/fparser.hh
[ 20%] Completed 'fparser'
[ 20%] Built target fparser
[ 22%] Creating directories for 'CSXCAD'
[ 25%] No download step for 'CSXCAD'
[ 27%] No update step for 'CSXCAD'
[ 30%] No patch step for 'CSXCAD'
[ 32%] Performing configure step for 'CSXCAD'
[... 117 lines cut ...]
 | (C) 2010-2018 Thorsten Liebig <thorsten.liebig@gmx.de>  GPL license
 ---------------------------------------------------------------------- 
	Used external libraries:
		CSXCAD -- Version: v0.6.2
		hdf5   -- Version: 1.10.7
		          compiled against: HDF5 library version: 1.10.7
		tinyxml -- compiled against: 2.6.2
		fparser
		boost  -- compiled against: 1_74
		vtk -- Version: 7.1.1
		       compiled against: 7.1.1

```
Any ideas how to proceed? Maybe I have differing versions in the system and load the wrong one? I found openEMS and AppCSXCAD in /usr/bin and the newly installed in /foss/tools/openEMS/bin/, but the failure seems to stem from some python module. There seems to be only one place in /usr/local/lib/python3.10/dist-packages/ which contains freshly build
```
drwxr-xr-x  2 root root    4096 Dec 16 20:58 h5py.libs
drwxr-xr-x  4 root root    4096 Dec 16 20:58 h5py
drwxr-xr-x  2 root root    4096 Dec 16 20:58 h5py-3.12.1.dist-info
-rw-r--r--  1 root root 2001728 Dec 16 21:00 CSXCAD-0.6.2-py3.10-linux-x86_64.egg
-rw-r--r--  1 root root  502651 Dec 16 21:00 openEMS-0.0.36-py3.10-linux-x86_64.egg
drwxr-xr-x  1 root root    4096 Dec 16 21:01 .
drwxr-xr-x  5 root root    4096 Dec 16 21:01 Cython-3.1.0a1-py3.10-linux-x86_64.egg
-rw-r--r--  1 root root     121 Dec 16 21:01 easy-install.pth

```

SakuraWutk, 2025-03-06:
I encountered the same error. 
I am using the ubuntu 20.04 in a server. and I install an anaconda with python 3.12.7