thliebig/openEMS-Project #235: header files not found

AutoEM, 2024-10-13:
hi,

got the following error:
```
C:\openEMS-Project\CSXCAD\python\CSXCAD\CSPrimitives.cpp(1257): fatal error C1083: Cannot open include file: 'CSXCAD/ParameterObjects.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit code 2
```

ParameterObjects.h is in 'CSXCAD\src' folder. How would I change this thank you?

karinemariew, 2024-10-20:
I assume you're doing the installation from the guide. I had this issue with the formatting of the -Include parameter in the following command, as listed on the installation guide.

**Original**: python setup.py build_ext -IC:\opt\openEMS\include -LC:\opt\openEMS

**Altered Version**: python setup.py build_ext -I "C:\opt\openEMS\include" -L "C:\opt\openEMS"

And replace opt with actual filepath

SengerM, 2024-11-03:
#239 may also fix this.