thliebig/openEMS-Project #241: Python Bindings fails on Mac M2

pkeshava, 2024-10-22:
I've successfully used brew to install openems and have verified that AppCSXCAD runs. However, I cannot get the python bindings to install. I've tried many things including the recommended steps in the documentation along with the discussion in #190 with no luck. The main issue is that the linker cannot find the CSXCAD lib, i.e., regardless of whatever method I try, I eventually get the error:

ld: library 'CSXCAD' not found
      clang++: error: linker command failed with exit code 1 (use -v to see invocation)
      error: command '/usr/bin/clang++' failed with exit code 1

after I've run

```shell
/opt/homebrew/opt/python@3.13/bin/pip3 install python/. --break-system-packages --user
```

I suppose I have to use LDFLAGS=...? but I'm not sure of the path to the lib??
This error occurs regardless of whether I use a virtual env python or the system wide as shown above

SengerM, 2024-11-03:
May want to try #239 ? I was having several issues installing the Python modules on Linux, with that now it works like a charm.

pkeshava, 2024-11-03:
> May want to try #239 ? I was having several issues installing the Python modules on Linux, with that now it works like a charm.

That seems to have done the trick. Thank you!!

aWZHY0yQH81uOYvH, 2024-11-03:
Did Homebrew successfully install the python packages to its managed version of python? You shouldn't have to install it separately.

Try running `/opt/homebrew/bin/python3` and see if `import openEMS` works. To make a python environment you can install further packages into, I would create a venv from that version of python with the `--system-site-packages` option.