thliebig/openEMS-Project #243: About installing cp310 vs cp311

Basilisvirus, 2024-10-25:
Why under  `openEMS_v0.0.36\openEMS\python` there are both cp311 and cp310 versions available to install? I believe cp310 is old(er) version. Is there any reason to prefer cp310 over cp311?

[image]

The [documentation](https://docs.openems.de/python/install.html#python-windows-install) sets as example installing cp310. But does not state anything about cp311.

thliebig, 2024-12-01:
The only reason is that I do not want to force people to use a certain python version just because of openEMS. Nothing else...

closed 2024-12-01 by thliebig

Basilisvirus, 2024-12-01:
but some of the examples dont run with both versions. cp310 looks more compatible with the examples. Also, with cp10 I managed to run a electric/magnetic field simulation while with cp311 it showed weird results.

If I make an issue here on the weird results I got on my simulation with cp311 would I get any help here regarding the reason behing the weird results?

thliebig, 2024-12-01:
This is the first time I hear about issues like this. Both are linked against the very same openEMS.dll and thus this is surprising... Maybe you can state which examples? I currently have no Windows machine but maybe someone can confirm your finding... BTW I always build this using WinPython... Not sure if this matters...