thliebig/openEMS-Project #260: Compilation Fails with MPI

AbleArcher88, 2024-12-01:
[build_20241201_001450.log](https://github.com/user-attachments/files/17967992/build_20241201_001450.log)

When I attempt to build with MPI enabled I get the enclosed error. No problems with no MPI.

thliebig, 2024-12-01:
I simply recommend to stay without MPI...

AbleArcher88, 2024-12-01:
Understood. Do is there a known issue with MPI?

thliebig, 2024-12-01:
When I implemented the MPI the best consumer CPU's just had a single thread and this was an attempt to speed things up (long lone time ago).
But today even the memory bandwidth between CPU cores is the major bottleneck... surely not the machine interconnection.  The MPI interface is just never used and thus I'm not sure if it even works anymore or how to get it running and to be honest, nobody should really bother. Maybe we should just remove it...

AbleArcher88, 2024-12-01:
Okay, thanks for the information. I encourage you not to remove it; I think it is a good idea to keep; I just wanted to put this on the radar.

thliebig, 2026-09-16:
MPI was really broken and rotting for a very long time now. It has been removed for v0.37. Maybe someday it can be re-implemented in a better and more efficient way. As it was, it was never a feasible implementation for modern machines.

closed 2026-09-16 by thliebig