thliebig/openEMS-Project #283: `fparser.hh:14:10: fatal error: 'string' file not found` when installing on macos

galah92, 2025-01-14:
Macbook Air M1.

```
➜  ~ brew tap thliebig/openems https://github.com/thliebig/openEMS-Project.git
➜  ~ brew install --HEAD openems
==> Fetching thliebig/openems/openems
==> Cloning https://github.com/thliebig/openEMS-Project.git
Updating /Users/galah92/Library/Caches/Homebrew/openems--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at fe3d796 update CSXCAD & QCSXCAD submodules
Entering 'AppCSXCAD'
Entering 'CSXCAD'
Entering 'CTB'
Entering 'QCSXCAD'
Entering 'fparser'
Entering 'hyp2mat'
Entering 'openEMS'
/Users/galah92/Library/Caches/Homebrew/openems--git/AppCSXCAD
/Users/galah92/Library/Caches/Homebrew/openems--git/CSXCAD
/Users/galah92/Library/Caches/Homebrew/openems--git/CTB
/Users/galah92/Library/Caches/Homebrew/openems--git/QCSXCAD
/Users/galah92/Library/Caches/Homebrew/openems--git/fparser
/Users/galah92/Library/Caches/Homebrew/openems--git/hyp2mat
/Users/galah92/Library/Caches/Homebrew/openems--git/openEMS
==> Installing openems from thliebig/openems
==> cmake .
==> make
Last 15 lines from /Users/galah92/Library/Logs/Homebrew/openems/02.make:
      |          ^~~~~~~~
In file included from /tmp/openems-20250114-33512-l3j2n3/fparser/fpoptimizer.cc:21:
/tmp/openems-20250114-33512-l3j2n3/fparser/fparser.hh:14:10: fatal error: 'string' file not found
   14 | #include <string>
      |          ^~~~~~~~
1 error generated.
make[5]: *** [CMakeFiles/fparser.dir/fparser.cc.o] Error 1
make[5]: *** Waiting for unfinished jobs....
1 error generated.
make[5]: *** [CMakeFiles/fparser.dir/fpoptimizer.cc.o] Error 1
make[4]: *** [CMakeFiles/fparser.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [fparser-prefix/src/fparser-stamp/fparser-build] Error 2
make[1]: *** [CMakeFiles/fparser.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so to (not Homebrew/brew or Homebrew/homebrew-core):
  thliebig/openems

Warning: Unable to query GitHub for recent issues on the tap
Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}]
openems's formula was built from an unstable upstream --HEAD.
This build failure is expected behaviour.
Do not create issues about this on Homebrew's GitHub repositories.
Any opened issues will be immediately closed without response.
Do not ask for help from Homebrew or its maintainers on social media.
You may ask for help in Homebrew's discussions but are unlikely to receive a response.
Try to figure out the problem yourself and submit a fix as a pull request.
We will review it but may or may not accept it.
```

Am I missing something? Would appreciate any help.

SakuraWutk, 2025-03-06:
I have the same error in the linux server

thliebig, 2026-09-16:
Should no longer be an active path. Re-open if still an issue

closed 2026-09-16 by thliebig