thliebig/openEMS-Project #284: AssertionError when trying to run example python file

aagum-bae, 2025-01-20:
These are the steps I followed for installation on windows:
1. Install python 3.10
2. pip install matplotlib, h5py
3. install the wheel files version 310
4. tried to run the example files to see if everything has been setup properly and i get the following error:

`= RESTART: C:\openEMS_v0.0.36\openEMS\python\Tutorials\Simple_Patch_Antenna.py =
Traceback (most recent call last):
  File "C:\openEMS_v0.0.36\openEMS\python\Tutorials\Simple_Patch_Antenna.py", line 114, in <module>
    FDTD.Run(Sim_Path, verbose=3, cleanup=True)
  File "openEMS\openEMS.pyx", line 489, in openEMS.openEMS.openEMS.Run
AssertionError
`

mdkendall, 2025-02-22:
The assertion that is failing is `assert os.getcwd() == os.path.realpath(sim_path)`. It was failing for me because `os.getcwd()` returned `C:\\Users\\MDKEND~1.PER\\AppData\\Local\\Temp\\Simp_Patch` whereas `os.path.realpath(sim_path)` returned `C:\\Users\\mdkendall.PERIODIC\\AppData\\Local\\Temp\\Simp_Patch`. The difference being that in one the username in the path is truncated to the "short" equivalent.

Note that this is on a domain-joined Windows PC, which is why the username portion of the home directory path is of the form `user.DOMAIN`.

I worked around this by modifying the example script `Simple_Patch_Antenna.py` to set `Sim_Path` using `os.getcwd()` instead of `tempfile.gettempdir()`, which seemed to work. The result is that the output files are in a folder in the current working directory, rather than a temporary directory.

aagum-bae, 2025-05-08:
I am trying to simulate a pcb after exporting via freecadtoems. And I am running into same assertino error