thliebig/openEMS-Project #316: cmake fails when build attempted under linux

StefanRastocky, 2025-04-18:
Hey, there seems to be a cmake version incompatibility - the project has about 10 or so CMakeLists.txt files, and each of them specifies cmake_minimum_required() version of 2.8 or 3.0, however, the versions of cmake after 4.0 consider this a problem and the build fails. Can you please update the source to an up to date minimum cmake version?

thanks

thliebig, 2025-04-18:
Yes I'm aware of this and was very angry with cmake. Because the idea is that this can be compiled on very old machines too.
but how to solve this? What would be a good minimum required? And why is cmake unable to be backward compatible as I would strongly suggest a tool like cmake should really try as hard as possible to be???

thliebig, 2025-04-18:
I find it completely stupid that you define a minimum required version of cmake like 3.0 and cmake errors out when the version is 4. That is just purely insane... Or what is the idea behind it that I do not understand?

thliebig, 2025-04-18:
I guess we just add the `-DCMAKE_POLICY_VERSION_MINIMUM=3.5` line everywhere?

thliebig, 2025-04-18:
What also works is an environment variable:
`export CMAKE_POLICY_VERSION_MINIMUM=3.5`

StefanRastocky, 2025-04-18:
Yeah, it is weird that cmake would abandon backwards compatibility with systems that are likely to still be used in many settings. Probably should keep their systems updated though :D 

Only explanation I can come up with is that something actually changed, that makes the cmake versions incompatible? but that's just a guess, I didn't look into it much yet - just noticed I cannot build openEMS. Thanks for pushing a commit, will look if it compiles now

StefanRastocky, 2025-04-20:
OK, still fails to compile, for a related issue, but the immediate problem with cmake version was removed. Here is the fail, I think you are trying to force "OLD" behaviour for window qt linking, so QCXCAD fails to configure:

> CMake Error at CMakeLists.txt:15 (cmake_policy):
>   Policy CMP0020 may not be set to OLD behavior because this version of CMake
>   no longer supports it. The policy was introduced in CMake version 2.8.11,
>   and use of NEW behavior is now required.

will it be fatal if i completely remove the CMP0020 line?

StefanRastocky, 2025-04-20:
OK, now it compiles. The necessary edits (sorry I didn't make a diff for a patch):

in QCXCAD CMakeLists.txt the "OLD"s were changed to "NEW", and also module for C language was not invoked, so that also failed the build, this needed a change mentioning C explicitely, which was ommitted before:

> project(QCSXCAD LANGUAGES C CXX)
> find_package(MPI COMPONENTS C CXX REQUIRED)

with that it builds nicely using an up to date cmake

thliebig, 2025-04-21:
Yes, sorry I found those issues too but forget to commit and push them.
I removed the policy lines since I believe "NEW" is default anyhow? I think this well get and bite me once I try to build on Windows...

thliebig, 2025-04-21:
Uhm no, I did not forget. Please check if the latest version and updated submodules build for you as they were please.

StefanRastocky, 2025-04-21:
after your latest commit 794a5d8 i pulled and tried to build, but run into the MPI error and the policy error, hence the pull request

closed 2025-04-21 by StefanRastocky