thliebig/openEMS-Project #319: Non-linear lumped components

BatchDrake, 2025-04-26:
Hi,

I don't expect this question to be answered in a straight-forward manner, but...

For a _thing_ I am playing with, I would like to simulate a PCB with a varactor diode somewhere. I understand this diode quite well (it is a SMV1231 if you are curious) and I've also written a SPICE model for it with success. Now, according to the documentation, adding non-linear lumped components seems unsupported for the time being.

Let us say I wanted to write support for non-linear lumped components to OpenEMS. Has this ever been considered? How many things I would need to fiddle with?

VolkerMuehlhaus, 2025-05-13:
From a simulation workflow perspective, it is usually better to split the design into layout for EM (passive, linear) and lumped models (linear or non-linear). These can then be combined using circuit simulation. 

The advantage is that you can vary all the excitations in circuit simulation, without doing the EM-simulation again and again. The layout part is only simulated once in EM because it's totally independent of circuit simulation levels.

I understand that adding ports for the lumped models means multiple FDTD excitations, so for linear RLC it's totally fine to include them as a part of the EM Model. And indeed, there are EM tools out there which could include a nonlinear SPICE model instead of placing circuit ports, but to me that looks like a bullet item in the feature list rather than a useful feature. You would need to excite the EM model with the exact waveform at the exact levels ...

BatchDrake, 2025-05-13:
Correct me if I am wrong: what you suggest is to take this linear part (the PCB layout basically, because I have no RLCs whatsoever), place the appropriate ports everywhere they are needed (e.g. actual circuit ports + where I would place my varactor diode), simulate, guess the S parameters and model the PCB as a frequency-dependent black box inside a SPICE circuit simulation?

If it is *this easy* I am going to feel a bit dumb now.

VolkerMuehlhaus, 2025-05-13:
Yes, your understanding is correct. That's the common workflow for such design tasks. 

To model the nonlinear effects properly, you need to cover DC up to some harmonics in your black box S-parameters. I suggest to set upper frequency limit as 5x target frequency or so, and DC as lower frequency limit.

closed 2025-05-13 by BatchDrake