thliebig/openEMS-Project #324: CSXCAD.CSPrimitives has no attribute POINT

Campbell-Rankine, 2025-05-14:
python=3.11 but also doesn't work when I change to 3.10

Full error message:

```
Traceback (most recent call last):
File "/usr/local/lib/python3.10/dist-packages/rq/worker.py", line 1644, in perform_job
return_value = job.perform()
File "/usr/local/lib/python3.10/dist-packages/rq/job.py", line 1344, in perform
    self._result = self._execute()
File "/usr/local/lib/python3.10/dist-packages/rq/job.py", line 1378, in _execute
    result = self.func(*self.args, **self.kwargs)
 File "/usr/src/app/openems/./worker.py", line 36, in worker_fn
    raise e
File "/usr/src/app/openems/./worker.py", line 32, in worker_fn
    result = patch.simulate(shape_opt=data.shape_opt)
File "/usr/src/app/openems/./patch/solver.py", line 40, in simulate
    return simulate_antenna(
File "/usr/src/app/openems/./patch/modules/simulate_openEMS.py", line 221, in simulate_antenna
    FDTD.AddEdges2Grid(
File "openEMS/openEMS.pyx", line 457, in openEMS.openEMS.openEMS.AddEdges2Grid
File "/usr/local/lib/python3.10/dist-packages/openEMS-0.0.36-py3.10-linux-x86_64.egg/openEMS/automesh.py", line 29, in mesh_hint_from_primitive
    if primitive.GetType() == CSPrimitives.POINT:
AttributeError: module 'CSXCAD.CSPrimitives' has no attribute 'POINT'
```

dodaduedada, 2025-05-22:
Hi,
i have the same message "AttributeError: module 'CSXCAD.CSPrimitives' has no attribute 'POINT'"" (using python 3.12).
Compiled everything from master:
 ----------------------------------------------------------------------
 | openEMS 64bit -- version v0.0.36-76-gc9d1d31
 | (C) 2010-2023 Thorsten Liebig <thorsten.liebig@gmx.de>  GPL license
 ----------------------------------------------------------------------
        Used external libraries:
                CSXCAD -- Version: v0.6.3-37-g3314deb
                hdf5   -- Version: 1.10.10
                          compiled against: HDF5 library version: 1.10.10
                tinyxml -- compiled against: 2.6.2
                fparser
                boost  -- compiled against: 1_83
                vtk -- Version: 9.1.0
                       compiled against: 9.1.0

CSXCAD/pyhton/setup.py shows version = '0.6.2' which is probably not matching installed CSXCAD version 0.6.3.
Windows openEMS binary 0.0.36 is working properly with .\openEMS_v0.0.36\openEMS\python\CSXCAD-0.6.3-cp311-cp311-win_amd64.whl

Error message in linux, executing tutorial file "simple_patch_antenna_v01.py":
"~/openEMS-Project$ /home/shadow/openEMS-Project/venv/bin/python /home/shadow/openEMS-Project/simple_patch_antenna_v01.py
/home/shadow/openEMS-Project/simple_patch_antenna_v01.py:139: SyntaxWarning: invalid escape sequence '\R'
  plot(f/1e9, np.real(Zin), 'k-', linewwidth=2, label = '$\Re\{Z_{in}\}$')
/home/shadow/openEMS-Project/simple_patch_antenna_v01.py:140: SyntaxWarning: invalid escape sequence '\I'
  plot(f/1e9, np.imag(Zin), 'r--', linewwidth=2, label = '$\Im\{Z_{in}\}$')
Traceback (most recent call last):
  File "/home/shadow/openEMS-Project/simple_patch_antenna_v01.py", line 65, in <module>
    FDTD.AddEdges2Grid(dirs='xy', properties=patch, metal_edge_res=mesh_res/2)
  File "openEMS/openEMS.pyx", line 457, in openEMS.openEMS.openEMS.AddEdges2Grid
  File "/home/shadow/openEMS-Project/venv/lib/python3.12/site-packages/openEMS/automesh.py", line 29, in mesh_hint_from_primitive
    if primitive.GetType() == CSPrimitives.POINT:
                              ^^^^^^^^^^^^^^^^^^
AttributeError: module 'CSXCAD.CSPrimitives' has no attribute 'POINT'"

thliebig, 2025-05-25:
Something in the latest cython must have caused this, because I do not see or remember any change that could cause this enums to not be visible anymore... Not sure yet how to fix this...

thliebig, 2025-05-25:
I have fixed the automesh function. Not sure if more places have to be fixed/changes.

SengerM, 2025-07-08:
Today I got this when trying to run the patch antenna example in Python.

thliebig, 2026-09-16:
This should be fixed in the latest master, see [here](https://github.com/thliebig/openEMS/commit/3f40de418d275d7ecdb8e3b56d24fb4f9d2ef5f7)

closed 2026-09-16 by thliebig