thliebig/openEMS-Project #328: Brew installation on Mac M4 - Error: tinyxml has been disabled because it is deprecated upstream!

VolkerMuehlhaus, 2025-06-05:
I am trying to set up on Mac M4, MacOS 15.5

Installation fails here, while processing brew install --HEAD openems
Any ideas?

```
==> Fetching vtk
==> Downloading https://ghcr.io/v2/homebrew/core/vtk/manifests/9.4.2_1
######################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/vtk/blobs/sha256:f547aa7af3480a
######################################################################### 100.0%
Error: tinyxml has been disabled because it is deprecated upstream! It was disabled on 2025-06-03.
```

aWZHY0yQH81uOYvH, 2025-06-06:
Looks like there's a new version of tinyxml (tinyxml2) which would need to be supported by openEMS. Trying to compile with only tinyxml2 does not work since CMake doesn't recognize it. Likely some API changes so may not be entirely trivial to make it work (?). In the short term, you can try to force Homebrew to install the old tinyxml. I think this would require something along the lines of invoking brew with `HOMEBREW_NO_INSTALL_FROM_API=1`, downloading ("tapping") the default formulae (`homebrew-core`), and manually removing [the offending line](https://github.com/Homebrew/homebrew-core/blob/6b6b9022399dd75e6563db0c3d8bb81433e59625/Formula/t/tinyxml.rb#L28) from the formula file.

VolkerMuehlhaus, 2025-06-06:
I tried to force installation of tinyxml (only this) with support from chatgpt, but ran into another error. This is what I tried:

```
mkdir -p ~/homebrew-tap/Formula
cd ~/homebrew-tap/Formula
```

The downloaded https://github.com/Homebrew/homebrew-core/blob/master/Formula/t/tinyxml.rb to that directory and used TextEditor to comment out the line that starts with disable!, finally save file

```
brew tap-new my/tap
brew install --build-from-source ./tinyxml.rb

```
Result: brew does several steps, but then stops with a cmake error complaining about cmake_minimum_required, Compatibility with CMake < 3.5 has been removed from CMake

I'll give up for now, and keep using openEMS on other platforms. Would have been nice to see how it performs on M4, but it seems more difficult than expected.

aWZHY0yQH81uOYvH, 2025-06-06:
Yep I ran into that same error before when I was testing before posting my previous comment. I couldn't remember if I'd run into that before and if Homebrew took care of it so didn't mention it.

It is definitely possible to get it working on M4, it's just that Homebrew is a "rolling release" package manager, meaning it always uses the latest versions, so with software written quite a while ago like openEMS, things break often. Nothing really to do with the OS/architecture.

Bypassing that CMake error is probably a matter of just updating the version the CMakeLists file asks for (likely won't cause any problems) or passing some command line option (I think printed with the error it gives?) to CMake to bypass the check.

I may make a PR at some point that fixes the CMake issue since that should be easy. I don't know how difficult it would be to switch tinyxml versions.

aWZHY0yQH81uOYvH, 2025-06-07:
Yeah just had a look at it and tinyxml2 uses a completely different memory management model which would probably require significant modifications to CSXCAD.

biergaizi, 2025-06-21:
In my opinion, since it's not practical to port CSXCAD to TinyXML in the short term, I believe the best stop-gap solution is to ask @aWZHY0yQH81uOYvH to create a fork the TinyXML Homebrew package in a personal repo, which can then be pulled by openEMS as a dependency.

In the long term, moving to TinyXML 2 is the eventual solution. Right now, only Homebrew disabled this package, other systems are still providing it, but it can't be guaranteed in the future. TinyXML is no longer updated after 2011, so it's considered obsolete even on Ubuntu 14.04. Now it has multiple unfixed security vulnerability (see [unofficial patches](https://gitlab.alpinelinux.org/alpine/aports/-/tree/master/community/tinyxml), which should be included in the fork). But due to API differences, it's not a trivial port.

aWZHY0yQH81uOYvH, 2025-06-21:
I suppose I could make a fork of the TinyXML repo and update the CMake compatibility version in all the OpenEMS components since the version requested is so old CMake no longer supports it. However, I've spent far longer fixing the Homebrew thing than actually using OpenEMS so I can't quite justify the likely half a day I'd have to spend fighting with it.

biergaizi, 2025-06-21:
I can fix everything else except the Homebrew package today on macOS today. The CMake should be a easy fix that I can do. Can you "simply" fork the `.rb` file for TinyXML 1 in your personal repo to allow pulling this dependency in Homebrew?

aWZHY0yQH81uOYvH, 2025-06-21:
I might try some time in the next few days if I get bored :)

biergaizi, 2025-07-11:
> but as a user, I have no idea how to apply this. It would be great to have a functional homebrew package.

As stated, the Homebrew package is maintained solely by aWZHY0yQH81uOYvH up to the moment. I don't use macOS, and I'm not sure if I have enough attention to pick up even more tasks, namely spending time and effort to learn a packaging system for an operating system that I don't use.

The best solution as of now is to retire the package and replace it with updated documentation. Perhaps the TinyXML installation can be added into ` ./update_openEMS.sh` for macOS as a special case.

---

This is my documentation draft, to be submitted to the upstream:

### macOS ###

openEMS can be installed via a Homebrew formula. However, the macOS package may not be up-to-date and can contain known problems. Often it's necessary to build your own development version, if so, follow this guide.

#### Warning ####

As of writing, the Homebrew formula for macOS is broken. The following manual installation is *required*!

#### Requirement ####

- openEMS depends on the following packages for minimum functionality:
  
      brew install cmake boost hdf5 cgal vtk

- Unfortunately, TinyXML has been disabled by Homebrew at has been unmaintained since 2011, TinyXML2 is not API-compatible. As a stop-gap, we have to build it from source. First, we obtain the last available version of TinyXML.
  
      # -L: follow redirect, REQUIRED!
      curl -L https://sourceforge.net/projects/tinyxml/files/tinyxml/2.6.2/tinyxml_2_6_2.tar.gz -o tinyxml-2.6.2.tar.gz
      tar -xf tinyxml-2.6.2.tar.gz
      cd tinyxml

- Next, we patch TinyXML to fix several known compatibility and security vulnerability. These patches came from various sources, and are applied by Homebrew, Debian, Alpine by their respective package maintainers. Here, we choose patches as maintained by AlpineLinux.
  
      # The first patch enforces use of stl strings, rather than a custom string type.
      # The second patch is a fix for incorrect encoding of elements with special characters
      # The third and fourth patches are security fixes.
      #
      # -L: follow redirect, REQUIRED!
      # -O: save to disk with an automatic file name.
      curl -L -O "https://raw.githubusercontent.com/alpinelinux/aports/b1ff376e83eb49c0127b039b3684eccdf9a60694/community/tinyxml/tinyxml-2.6.2-defineSTL.patch"
      curl -L -O "https://raw.githubusercontent.com/alpinelinux/aports/b1ff376e83eb49c0127b039b3684eccdf9a60694/community/tinyxml/tinyxml-2.6.1-entity.patch"
      curl -L -O "https://raw.githubusercontent.com/alpinelinux/aports/b1ff376e83eb49c0127b039b3684eccdf9a60694/community/tinyxml/CVE-2021-42260.patch"
      curl -L -O "https://raw.githubusercontent.com/alpinelinux/aports/b1ff376e83eb49c0127b039b3684eccdf9a60694/community/tinyxml/CVE-2023-34194.patch"
      
      patch -p1 < tinyxml-2.6.2-defineSTL.patch
      patch -p1 < tinyxml-2.6.1-entity.patch
      patch -p1 < CVE-2021-42260.patch
      patch -p1 < CVE-2023-34194.patch

- Then, we introduce CMake support to TinyXML:
  
      # The final patch adds a CMakeLists.txt file to build a shared library and provide an install target
      # submitted upstream as https://sourceforge.net/p/tinyxml/patches/66/
      curl -L -O "https://gist.githubusercontent.com/scpeters/6325123/raw/cfb079be67997cb19a1aee60449714a1dedefed5/tinyxml_CMakeLists.patch"
      
      # You know something is truly deprecated when the patch itself needs
      # patching! In CMake 4, 3.10 is deprecated and 3.5 has been removed.
      # Replace "cmake_minimum_required(VERSION 2.4.6)" in the patch with
      # "cmake_minimum_required(VERSION 3.0...3.10)".
      sed -i -e "s/cmake_minimum_required(VERSION 2.4.6)/cmake_minimum_required(VERSION 3.0...3.10)/" \
                tinyxml_CMakeLists.patch  # -e is not optional in BSD sed
      patch -p1 < tinyxml_CMakeLists.patch

- Finally, TinyXML can be installed to a custom user directory. Here, we use ``$HOME/opt/openEMS`` as an example. This directory must match the directory later used for installing openEMS.
    
      mkdir build && cd build
      cmake ../ -DCMAKE_INSTALL_PREFIX=$HOME/opt/openEMS
      make && make install

- To use Paraview to visualize simulation results (recommended):
[... 47 lines cut ...]
  * ``--with-hyp2mat``: Enable hyp2mat for HyperLynx PCB layouts (unmaintained, not recommended).
  * ``--with-MPI``: Build MPI engine (only needed for cluster, not recommended).
  * ``--disable-GUI``: Disable AppCSXCAD GUI, useful for servers.

For example, to build openEMS with CTB and python:

    bash ./update_openEMS.sh ~/opt/openEMS --with-CTB --python

#### openEMS search path ####

After the build is complete, add `~/opt/openEMS/bin` into your search path::

    export PATH="$HOME/opt/openEMS/bin:$PATH"

You need to write this line into your shell's profile, such as `~/.bashrc` or `~/.zshrc` to make this change persistent.

#### Octave Search Path ####

Assuming that we've built openEMS from source and installed it into a prefix ``~/opt/openEMS``, the required Matlab/Octave libraries can be found at ``~/opt/openEMS/share/CSXCAD/matlab`` and ``~/opt/openEMS/share/openEMS/matlab``. You may add these folders manually into ``~/.octaverc`` via:

    % change the prefix ~/opt/openEMS to the path on your machine
    addpath('~/opt/openEMS/share/openEMS/matlab');
    addpath('~/opt/openEMS/share/CSXCAD/matlab');

For system-wide installation via root, these folders are usually located ``/usr/share/openEMS/matlab`` and ``/usr/share/CSXCAD/matlab``. Alternatively you can setup these path more permanently using the "Edit"-Menu using "Set Path" in Octave.

VolkerMuehlhaus, 2025-07-12:
> For example, to build openEMS with CTB and python:
> 
> ```
> bash ./update_openEMS.sh ~/opt/openEMS --with-CTB --python
> ```

In this step, I get an error:

```
Building python modules ... please wait
Python modules build failed, please see logfile for more details... /Users/volker/opt/openEMS/openEMS-Project/build_20250712_172213.log
```

The log is attached, besides lots of warnings there is this error at the very end:

```
In file included from /Users/volker/opt/openEMS/openEMS-Project/openEMS/python/openEMS/openEMS.cpp:1190:
/Users/volker/opt/openEMS/include/openEMS/openems.h:30:10: fatal error: 'boost/program_options.hpp' file not found
   30 | #include <boost/program_options.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
error: command '/usr/bin/clang++' failed with exit code 1
Python module build failed!
```

[build_20250712_172213.log](https://github.com/user-attachments/files/21197043/build_20250712_172213.log)

All help is much appreciated! 
cc @aWZHY0yQH81uOYvH

VolkerMuehlhaus, 2025-07-12:
I guess it needs to be 

`cd openEMS-Project/CSXCAD/python `

and 

`cd openEMS-Project/openEMS/python`

to locate the setup.py scripts.

I run into an error that I also experienced with the installation attempt above:

```
In file included from /Users/volker/opt/openEMS/openEMS-Project/openEMS/python/openEMS/openEMS.cpp:1190:
/Users/volker/opt/openEMS/include/openEMS/openems.h:30:10: fatal error: 'boost/program_options.hpp' file not found
   30 | #include <boost/program_options.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
error: command '/usr/bin/clang++' failed with exit code 1
```

Unfortunately, I have no ideas what that means and what I need to check.

biergaizi, 2025-07-12:
Don't copy and paste blindly, the fact that you copied the example without changing its content suggests you missed critical information contained within this paragraph.

> Tip: If Homebrew is used on macOS, -I$(brew --prefix)/include -L$(brew --prefix)/lib must be listed in additional to the openEMS prefix path, if additional paths such as $HOME/opt/include are already included, use comma as separator. This path can be different from /usr/local. Run brew --prefix to check, or keep $(brew --prefix) as placeholder.

How to apply this information is trivial, and is left as an exercise to the reader.

biergaizi, 2025-07-12:
> Installed with no errors now, but it seems that despite my openEMS venv being activated before the all the build and setup procedure, the openEMS Python interface was not installed in that venv?

This is an oversight in the documentation. The option `--user` should be removed if venv is used, otherwise it always attempts to install into the user's home directory. Thanks for beta-testing the documentation.

VolkerMuehlhaus, 2025-07-12:
By the way ... the simulation speed of Apple Silicon is impressive, very happy to get this up and running with your support.

For a simple test case from my RFIC workflow, speed using MacBook Air M4 was equal to AMD EPYC 7313P running v0.0.36-82-g8a008a4 on Ubuntu 24.04.

tdunning, 2025-07-25:
> 
> ```
> cd openEMS-Project/CSXCAD 
> 
> python3 setup.py build_ext \
>   -I "$HOME/opt/include:/usr/local/include" \
>   -L "$HOME/opt/lib:/usr/local/lib" \
>   -R $HOME/opt/lib
> 
> python setup.py install --user
> 
> cd openEMS-Project/openEMS
> 
> python3 setup.py build_ext \
>   -I "$HOME/opt/include:/usr/local/include" \
>   -L "$HOME/opt/lib:/usr/local/lib" \
>   -R $HOME/opt/lib
> 
> python setup.py install --user
> ```

Small nit here. You seem to alternate between `python3` and `python`. It should all be `python3`, I think. If it isn't, it deserves a note.

biergaizi, 2025-10-21:
@Niklas-2911 This is a very new bug that didn't exist when I was writing this. It was introduced by @Aloz1 in PR #314, who appeared to believe that GNU/Linux is the only Unix-like operating system in existence, and used the GNU coreutils specific command `nproc` in the script. I also have responsibility, since my CI/CD script clearly detected the bug, but instead treated it as a non-fatal error: https://github.com/thliebig/openEMS-Project/actions/runs/18499675059/job/52712846432#step:4:22

I believe GNU commands are available on macOS so you can just:

     brew install coreutils

without changing the script.

Alternatively, for people who don't want to install third-party packages, change line 36 of `update_openEMS.sh` from 

    NJOBS=$(nproc)

to

    NJOBS=8

We really need a full-time developer to just pay attention to the build matrix. Otherwise, it will break repeatedly whenever I'm not looking at it.

biergaizi, 2025-10-21:
Also, it appears that in Boost 1.89, released a few weeks ago, the module `system-config` has been *removed* because it's header only, so it's pointless to check anything. As a result, the Boost check fails:

```
CMake Error at /opt/homebrew/lib/cmake/Boost-1.89.0/BoostConfig.cmake:141 (find_package):
  Could not find a package configuration file provided by "boost_system"
  (requested version 1.89.0) with any of the following names:

    boost_systemConfig.cmake
    boost_system-config.cmake
```

To fix it, remove all mentions of `system` from `find_package()` in `CMakeLists.txt`, `CSXCAD/CMakeLists.txt`, `openEMS/CMakeLists.txt`, `QCSXCAD/CMakeLists.txt`, `insert more packages here`, etc.

For example, if you have

```
# boost
find_package(Boost 1.46 COMPONENTS
  thread
  system
  date_time
  serialization
  chrono
  program_options
  REQUIRED
)
```

Then remove `system`.

Note to all developers: for backward compatibility, this change must *not* be committed into the upstream, only locally. A proper fix as an if/else check based on Boost version.

thliebig, 2025-10-22:
@tdunning I agree with you. That comment was unnecessary harsh. Please @biergaizi, choose your wording a bit more careful. Thanks

I will review and merge this changes as soon as possible and thus hopefully be able to close this for good.