thliebig/openEMS-Project #332: Homebrew Installation Fails on M1 Mac: tinyxml deprecated (June 3, 2025)

vsmark42, 2025-06-15:
Hi OpenEMS Team / Community,

I'm trying to install openEMS on my M1 Mac following the official Homebrew instructions, but I'm encountering a critical error related to a tinyXML deprecated dependency.

Error Message:

Error: tinyxml has been disabled because it is deprecated upstream! It was disabled on 2025-06-03.

Context:
Operating System: macOS Sonoma (or specify your exact macOS version if different)
Processor: Apple M1 (ARM64)
Homebrew: Latest version (brew update run recently)
Command used: brew install --HEAD thliebig/openems/openems (after brew tap thliebig/openems https://github.com/thliebig/openEMS-Project.git)

What I've Tried:
I understand tinyxml is deprecated upstream. As a temporary workaround, I attempted to manually edit the tinyxml.rb formula in my local Homebrew core tap (/opt/homebrew/Library/Taps/homebrew/homebrew-core/Formula/tinyxml.rb) to comment out the disable! line, and then ran the installation with HOMEBREW_NO_INSTALL_FROM_API=1. But it's a fragile, unsupported workaround and not a sustainable solution.

biergaizi, 2025-06-21:
The macOS Homebrew port is maintained exclusively to @aWZHY0yQH81uOYvH, so CC @aWZHY0yQH81uOYvH here for their attention. None of us understands except them the Homebrew package system (although I'm familiar with bare-metal POSIX on macOS), so they have the final words on Homebrew support here.

In my opinion, since it's not practical to port CSXCAD to TinyXML in the shortterm, I believe the best stop-gap solution is to ask @aWZHY0yQH81uOYvH to create a personal fork the TinyXML Homebrew package in a personal repo, which can then be pulled by openEMS as a dependency.

In the long term, moving to TinyXML 2 is the eventual solution. Right now, only Homebrew disabled this package, other systems are still providing it, but it can't be guaranteed in the future. TinyXML is no longer updated after 2011, and including multiple security vulnerability (see [unofficial patches](https://gitlab.alpinelinux.org/alpine/aports/-/tree/master/community/tinyxml), which should be included in the fork). But due to API differences, it's not a trivial port.

biergaizi, 2025-06-21:
Deprecate of #328. Please close it in favor of #328.

aWZHY0yQH81uOYvH, 2025-06-21:
Duplicate of #328. I suppose I could make a fork of the TinyXML repo and update the CMake compatibility version in all the OpenEMS components since the version requested is so old CMake no longer supports it. However, I've spent far longer fixing the Homebrew thing than actually using OpenEMS so I can't quite justify the likely half a day I'd have to spend fighting with it.

biergaizi, 2025-06-21:
@aWZHY0yQH81uOYvH Let's continue this discussion at #328.