thliebig/openEMS-Project #335: DD26F00]vtkOpenGLPolyDataMapper:328   WARN| vtkOpenGLPolyDataMapper::SetGeometryShaderCode was deprecated for VTK 9.0

VolkerMuehlhaus, 2025-06-16:
Dear experts in CSXCAD,

In my automated RFIC EM workflow, valuable mesh information details are printed to the terminal, just before launching appCSXCAD.

However, on most recent Linux machines, the user will miss that information disappears because AppCSXCAD throws thousands of the these messages to the terminal:

`DD26F00]vtkOpenGLPolyDataMapper:328   WARN| vtkOpenGLPolyDataMapper::SetGeometryShaderCode was deprecated for VTK 9.0 and will be removed in a future version.  Use vtkOpenGLShaderProperty::SetGeometryShaderCode instead.`

Is there any chance to fix this? All help is much appreciated!

Best regards
Volker

VolkerMuehlhaus, 2025-06-16:
My ugly workaround at the moment: Instead of calling AppCSXCAD this way

`ret = os.system(AppCSXCAD_BIN + ' "{}"'.format(CSX_file))`

the call is modified by appending 2>/dev/nul, so that errors and warnings are redirected to nowhere:
`
ret = os.system(AppCSXCAD_BIN + ' "{}"'.format(CSX_file) + ' 2>/dev/null')`

thliebig, 2025-06-16:
Are you using the latest version? Or the last release? Because I use an even newer vtk and do not see this messages with the latest openEMS/AppCSXCAD

VolkerMuehlhaus, 2025-06-16:
> Are you using the latest version?

The machine where I run just now uses 
CSXCAD -- Version: v0.6.3-36-g038c99a
but it appears on every openEMS Linux installation that I have seen so far.

Another machine where the issue is the same:
CSXCAD -- Version: v0.6.3-37-g3314deb

VolkerMuehlhaus, 2025-06-16:
The issue was also adressed here, but I don't understand 
- how to apply that to my existing installations
- what that possibly means for published openEMS installation instructions

https://github.com/thliebig/openEMS-Project/discussions/137

thehans, 2025-08-16:
I'm seeing this log spam also in Linux Mint 22.1 (Ubuntu 24.04 base)
Where the version of vtk9 is `9.1.0` , so doesn't seem to be fixed in 9.1.0 at least.

I've built openEMS-project from github sources on 8/12

thliebig, 2026-09-16:
It is fixed only in >=vtk9.2 unfortunately. A workaround for 9.1+9.2 is implemented

closed 2026-09-16 by thliebig